AstarVienna/ScopeSim #13: Scaling of atmospheric background emission doesn"t work

marcogullieuszik, 2019-11-28:
Fix it

hugobuddel, 2023-03-22:
Sorry for the late response. I'm going through all the issues again to see whether something has been missed.

It seems that this issue has been resolved, because scaling of atmospheric background emission seems to work:

```
import astropy.units as u
from scopesim.effects.ter_curves import AtmosphericTERCurve

atmosphere1 = AtmosphericTERCurve(
    filename="./scopesim/tests/mocks/basic_instrument/ATMO_default_NIR_IMG.dat",
)
print(atmosphere1.emission(10000 * u.Angstrom))

atmosphere2 = AtmosphericTERCurve(
    filename="./scopesim/tests/mocks/basic_instrument/ATMO_default_NIR_IMG.dat",
    rescale_emission={
        "filter_name": "J",
        "value": 16.0,
        "unit": "mag",
    },
)

print(atmosphere2.emission(10000 * u.Angstrom))

atmosphere3 = AtmosphericTERCurve(
    filename="./scopesim/tests/mocks/basic_instrument/ATMO_default_NIR_IMG.dat",
    rescale_emission={
        "filter_name": "J",
        "value": 15.0,
        "unit": "mag",
    },
)
print(atmosphere3.emission(10000 * u.Angstrom))

print(
    atmosphere3.emission(10000 * u.Angstrom) / atmosphere2.emission(10000 * u.Angstrom)
)
```
results in
```
1.8309999999999278e-05 PHOTLAM
3.1124006061831195e-05 PHOTLAM
7.817996852093576e-05 PHOTLAM
2.511886431509582
```
It seems to scale by a factor of 2.5 for each magnitude, as expected.

I know it is long ago, but I did want to investigate this anyway. Shall we close this issue, or is there still something to be done?

carmeloarci, 2023-11-06:
The scaling of the atmospheric background seems still not properly addressed. 
[Test Background I.zip](https://github.com/AstarVienna/ScopeSim/files/13270233/Test.Background.I.zip)

hugobuddel, 2023-11-10:
As @teutoburg indicated in https://github.com/AstarVienna/irdb/issues/142#issuecomment-1806153433 , the Armazones package in the IRDB uses the `SkycalcTERCurve` for the atmosphere, which does not allow simple scaling to a specific magnitude (but only by actually changing the properties of the atmosphere).

It is unfortunate that the very first examples that people see (https://scopesim.readthedocs.io/en/latest/getting_started.html) actually contain scaling of the `AtmosphericTERCurve`, but without actually explaining that. So it is only natural that people assume they can scale the atmosphere for ELT observations as well, while they can't.

My suggestion is to remove the scaling of the atmosphere from that very first example. The default should be a physical atmosphere, like `SkycalcTERCurve`. The `AtmosphericTERCurve` should only be used if the `SkycalcTERCurve` is not sufficient, and should therefore be explained in a later section.

hugobuddel, 2023-11-10:
> The scaling of the atmospheric background seems still not properly addressed. [Test Background I.zip](https://github.com/AstarVienna/ScopeSim/files/13270233/Test.Background.I.zip)

This notebook is much more useful than the original bug report, but it would still not be enough for me to actually investigate what is going on. (Assuming I would not yet know, I believe the explanation in the previous comment to be correct.).

It is important to explicitly state a) what is expected and b) what you get. In this case, I would expect something like "in simulation 1 I set the background to X, in simulation 2 to Y, so I expect the background levels to be Z times higher in simulation 2, but as you can see from the numbers P, Q and R, the background level is identical".

As is, the notebook simply creates 20 simulations, and it works. So everything seems fine just going by the information presented in the notebook. I would probably have figured this one out though, but I might have postponed the investigation due to the uncertainty of what the report is actually claiming.

If I get a bug report with enough to chew on, that is, something that I can autonomously reproduce and test, then I'll happily (or unhappily) work till the wee hours in the night to figure out what is going on. Or maybe I can immediately see what is going on. That gets you the quickest return.