AstarVienna/ScopeSim #17: Problems with star positions

msim404, 2019-12-03:
Hi,
I found 2 issues when using source_templates.basic.stars:
1. x position is shifted by 7 px in the output image
2. x and y looks transposed

Here below the code to reproduce the problem.

```
import scopesim_templates
import scopesim as sim
from matplotlib import pyplot as plt
from astropy import units as u
import numpy as np

sim.server.database.download_package(["locations/Armazones.zip","telescopes/ELT.zip","instruments/MICADO.zip"])

cmds = sim.UserCommands(use_instrument="MICADO")

sp_types=['A0V', 'A0V']
mags=np.array([10., 10.])*u.mag
x=np.array([0., 0.])
y=np.array([0., -0.4])
star= scopesim_templates.basic.stars.stars('Ks',mags, sp_types,
                                           x,y)
srcs=star
cmds['!OBS.dit']=30
cmds['!OBS.ndit']=1
cmds['!SIM.random.seed']=None
opt=sim.OpticalTrain(cmds)
opt.observe(srcs)
hdu=opt.readout()[0]
fdat=hdu[1].data

xpix=x/0.004+512 + 7 # I have to add 7!!
ypix=y/0.004+512 

fig,ax=plt.subplots()
ims=ax.imshow(fdat,origin='lower',vmax=80000)
ax.plot(xpix,ypix,'xr')
fig.colorbar(ims)
ax.set_xlim(512-150,512+150)
ax.set_ylim(512-150,512+150)
fig.savefig('xy_test.png')

```

msim404, 2019-12-05:
[Update]
Point 1 was my fault: I was using the default airmass value (1.2).
Point 2 is still there, or am I missing something?

astronomyk, 2019-12-05:
Interesting that the atmospheric dispersion is still taken into account. Good to know.
Point 2 is very possible. I haven't yet put much time into the output FITS objects, but will keep this in mind when I get around to it.
In the meantime this should be easily correctable by using the transpose operator

    <hdu>[1].data.T

msim404, 2019-12-06:
Indeed, the result changing the filter in the simulation is (airmass=1.2): 
[image]
Do the ADC effect is implemented for MICADO?

hugobuddel, 2023-03-23:
I'm closing this issue since it seems the images are not transposed anymore.

However, during my on it seems that the ADC effect is broken, see #194.

closed 2023-03-23 by hugobuddel