AstarVienna/ScopeSim #22: ObservingBlock/observing Templates as input configuration files for ScopeSIM

gijsverdoeskleijn, 2020-06-07:


astronomyk, 2020-06-07:
Of course! One thing that may help will be the fortnightly ScopeSim-Pipeline telecons that @hugobuddel and I will set up from next week. 
Furthermore I'm thinking about doing a monthly ScopeSim videocon for the MICADO consortium for whomever is interested. Basically just an "Ask me anything" session, and if no-one has questions, I'll choose some section of the scopesim functionality and talk about that for an hour.

If you want to get started in a hands-on manner, I suggest looking at the [ScopeSim meets Astrometry](https://scopesim.readthedocs.io/en/latest/use_examples/jupyter_notebooks.html#adding-custom-effect-objects) ipython notebook, and understanding what's going on there with the "[Effects](https://github.com/astronomyk/ScopeSim/blob/master/scopesim/effects/effects.py)" plugin functionality. If you can wrap your head around whats going on there, then the rest of ScopeSim will become pretty clear.

astronomyk, 2020-06-07:
The [instrument reference database](https://github.com/astronomyk/irdb) can be found here. Have a look at the .yaml files in the MICADO package (mainly default.yaml and MICADO.yaml) to get an idea of how it will be possible to set up the joint scopesim/pipeline infrastructure

hugobuddel, 2020-06-11:
> * establishing the joint pipeline of ScopeSim and then Imaging pipeline. For which I have the dream that it is possible to have ObservingBlock/observingTemplates as input and delivers calibrated imaging as output.

This is essentially what we're already moving towards.

See the `make_simulation()` method in the `Raw` class: https://gitlab.astro-wise.org/micado/micadowise/-/blob/standardcado/src/micado/main/RawExtra.py#L55
Template and Observing Block instances are constructed there and passed on to ScopeSIM.

The resulting raws can subsequently be processed in a forward chaining matter into processed data. There is a stub method to make reduced Science data, but in reality this only works for Dark -> MasterDark yet.

Forward chaining might already be enough for your goals, but with backward chaining you should be able to directly request the processed data and have the raw data simulated and processed automatically.

hugobuddel, 2020-06-11:
> * establishing the joint instrument reference database, to be in common between ScopeSIM and Data Processing Subsystem (i.e., imaging pipeline, spectro pipeline, PSFR application pipeline)

We have that to, the IRBD that @astronomyk linked to is used in the data reduction prototype environment (MICADO-WISE).

At this moment the conversion between the IRDB (yaml files) and our digital design (vodsl) is manual. This will probably not change, because this works well enough. What we should do:
* Automatically verify that both models are consistent. We had this test, but it was disabled when it broke. See https://gitlab.astro-wise.org/micado/micadowise/-/issues/55
* Automatically use the defaults as provided by the IRDB.

I'm confident that this will work out well enough. There probably will be technical problems, but the design of ScopeSIM / IRDB / MICADO-WISE resemble each other well enough that I don't see any fundamental roadblocks.

gijsverdoeskleijn, 2020-06-16:
> Of course! One thing that may help will be the fortnightly ScopeSim-Pipeline telecons that @hugobuddel and I will set up from next week.

@astronomyk  and @hugobuddel Please put these in my agenda. I expect to join once per month / 6 weeks.

hugobuddel, 2020-06-17:
I think we should have one issue per topic, not two topics per issue. Because well, they are different issues.

I propose we use this topic to discuss Observing Blocks and Templates, and move the IRDB discussion elsewhere. But I don't see anything actionable or specific w.r.t. to the IRDB topic, so maybe we can drop it. If anything the ball is with MICADO-WISE to better integrate the IRDB, and there is already an issue for that. Any specific problem that arises there can be opened here if necessary. Is that okay with you @gijsverdoeskleijn.

As for the Observing Blocks / Templates, @gijsverdoeskleijn and me discussed this a bit this morning. We concluded that this should be our assumptions:

- An ObservingBlock contains several Templates, e.g. acquisition and observation template.
- Only one of those templates describes the actual science exposures we are interested in.
- This template has a number of exposures that usually correspond to the ditter steps of the observation.

So we can model them like that:
- The ObservingBlock provides information about the target (e.g. RA/DEC).
- The Template provides information like DET.NDIT, DET.DIT, FILT.NAME, and will also contain information like TPL.START.
- All other templates are ignored (and thus not modeled).

BTW, unlike I recalled this morning, an observing block sets specific parameters for each templates. (I'm not sure what I thought this morning, it cannot be another way.)