AstarVienna/ScopeSim #23: Kieran-Gijs todo list

gijsverdoeskleijn, 2020-06-16:
The Big Ones:
* establishing the joint instrument reference database
* establishing the joint pipeline of ScopeSim and then Imaging pipeline

The Small Ones:
* update to Science Team: monthly telecons: great idea
* WP Simulator roadmap update
* doc number for FDR ScopeSIM doc
* github+overleaf for FDR ETC doc
* Feedback on ScopeSIM paper: https://www.overleaf.com/2556697297ysnfqjjjdjv ,  https://mail.google.com/mail/u/0/?tab=rm#search/from%3Aleschinski/FMfcgxwDqxGNQcpmGGXkWJNlwrhXkMXs?compose=DfmMsHFtpZCTmmdSPpgGwdtCcBVzmLWXXdcTmbCgHFZZPsjFjfdfpfCVWtzhZKBJxwkfwBSXvvddBfgkhGLXJpZjsvJQJNFljDShfgWnpCTDlNQPTlwCcHvdMNHMstbBXtBNWfpnLLtNsrxjCRgQwLdzQmNlGnCMsLsdsDtHwnBnJkxbzRRsdxbDKkXhmgjnPFhcRmJBxtFCtsvBPkSDGLXrWzPcFmNjZfgbcLvjVnfvvlbnmBjv

gijsverdoeskleijn, 2020-06-23:
@hugobuddel and @astronomyk : could you bring already at my data flow splinter tomorrow the list of planned developments to allow simulation of raw data, calibration data and HCI mode for the imaging pipeline prototype. This is same request as in my note on Kieran's nice presentation:  https://docs.google.com/presentation/d/19iZOR4Brz7tMkFdILNjSeCbtzWwnS6k7lzLZm0nPwMg/edit?pli=1#slide=id.g8a950c5016_0_268 .

hugobuddel, 2020-06-23:
I'll create issues for features that are necessary to simulate the data

astronomyk, 2020-06-23:
@Hugo Buddelmeijer <buddel@astro.rug.nl>  could you also create an
"overview" issue that contains a list of everything you want

We can use the checklist markdown syntax and tick them off as we make them.
E.g:

- [x] Finish my changes
- [ ] Push my commits to GitHub
- [ ] Open a pull request

-------------------------------------
Kieran Leschinski
Department of Astrophysics
University of Vienna

hugobuddel, 2020-06-23:
I've added them as a list to this issue. We can make a new issue if this one becomes to large. For now this is manageable I think.

But otherwise I guess it would be more proper to create a milestone and assign the issues to that.

I believe that closed issues will be striked through (struck through?) (test #25 oh it does not). So the check boxes indeed make sense.

astronomyk, 2020-06-24:
Hi Gijs,

I wasn't sure where to put this, so for now it's here. That way Hugo can also chime in. I'll move these to separate issues after brain dumping here:

"... list of planned developments to faciliate simulation of raw data, calibration data and HCI mode.""

- raw data
  Raw data is easy to make already. (``.observe()``, ``.readout()``). What we need is a list of data that should be simulated. I guess that's the next point;
- calibration data
  Here I need a list of what calibration files you want. Then we need to write Source objects (5-10 lines max) for each of the calibration sources that would create these files. For example:
  - RON - super easy. Turn off everything execpt detector effects
  - Dark - easy, no Source needed, turn off atmo/tel thermal background. Keep Inst therm BG(?)
  - Flat - for a flat field source we need a map of the light distribution (parabolioc, flat) and a lamp spectrum
  - Pin hole image - need the position of the pin holes, a MICADO distortion map (if exists), and a lamp spectrum (or BB temp)
  For each of these we also need a YAML file containing the relevant commands for running the simulation
  I guess the next step is to brainstorm this list, unless you and Hugo already have the list somewhere
- HCI mode
  - Firstly I should chat with Pierre Baudoz about the current plan for the coronagraphs, and exactly what effects are involved, and whether or not they can be included in scopesim. My gut feeling is yes, they all can be. But a Frenchman will need to write the code for the relevant Effects objects. 
  - For any APP coronagraphs HCI mode should be easy. Just need a set of PSFs. Alternatively I can try to generate them myself with POPPY if I know the dimensions of the coronagraph masks. As they come after the WFS, they are simply an additional PSF Effect object that needs to be included
  - Vortex coronagraphs are going to be a bit trickier, as the PSF varies strongly as the point source moves within a tiny range. Hence small shakes from the AO system will likely cause wild PSF shapes over a single DIT.
  I am definitely not an expert here, hence why a good long chat with Pierre is necessary.

Hopefully that's enough for today. Lets have a chat in out DSF+SS telecon next week.

hugobuddel, 2020-06-24:
Thanks Kieran,
* What is RON?
* For the pin hole image: I'd like to create a specific issue for this somewhere, but not sure:
  * On ScopeSIM: does any effect class need to be written for these pin hole masks? (I guess yes)
  * On ScopeSIM_templates: can the pinhole mask be considered a source? (I guess no)
  * On IRDB: should the configuration of the pinhole mask be described there? (I guess yes)
  * On MICADO-WISE: there should be some hardware class describing the masks. (I know yes)
* For HCI I need to go through this soon.

For the rest see the issues I already created (the lists in this issue description).

astronomyk, 2020-06-24:
* On ScopeSIM_templates: can the pinhole mask be considered a source? (I guess no)
  The easy way would be here as the mask is a "quasi" source, it's simulating a grid of point sources

* On IRDB: should the configuration of the pinhole mask be described there? (I guess yes)
  The less easy way (but self-consistent) would be to make an Effect for the pinhole mask, and use the flat-field source object

hugobuddel, 2020-06-24:
Ok, then, let's go for the easy way: astronomyk/ScopeSim_Templates#6
If that fails, we can do the complicated way. Added to the list on top.

gijsverdoeskleijn, 2022-09-14:
@astronomyk Kieran:  could you create with ScopeSIM this raw science simulation: https://docs.google.com/presentation/d/1RhdKAbcDwxRQP8orjJVpILKm7Qd-h85m7x4GE23IWZw/edit#slide=id.g144ad46cf9f_0_31 and share the FITS files and scopesim configuration with @wjvriend and me? Then Willem-Jan can attempt to ingest it. And I can attempt to recreate it using the interface to ScopeSIM inside MicadoWISE.