AstarVienna/ScopeSim #24: realization of ScopeSIM archive

gijsverdoeskleijn, 2020-06-16:
HB says: In particular, how does this relate to our plans for a MICADO Archive?

I wasn't aware that we have committed to maintaining a MICADO archive. E.g. see the dbviewer link in datareductionsystem report.

In particular, do we plan to maintain the Oracle database behind the dbviewer? That would also require us to have a relatively stable data model. My plan up till now was to stop using that dbviewer, and perhaps even not use the data server, because they limit our capability of iterating quickly. Willem-Jan gave me control over those services, as I was also updating the code etc.

We should formulate a plan about what we plan to support, and for who.

What we perhaps can do is fully automate the generation of whatever data we want in this archive. then we can recreate all the data each time we merge new code into develop. Then we simply throw the entire database away and add all the new data. Otherwise it will be a pain to maintain all the different versions.

And for now we would probably still need to use an Oracle database, since not everything the dbviewer does is possible with PostgreSQL yet.

hugobuddel, 2020-06-17:
> HB says: In particular, how does this relate to our plans for a MICADO Archive?

In general I have our MICADO archive plans on low priority because it is not an FDR deliverable as far as I know.
 
> I wasn't aware that we have committed to maintaining a MICADO archive. E.g. see the dbviewer link in datareductionsystem report.

Only for the consortium right? And as far as I know this could just be a web page with hardcoded links to FITS files, which should be easy. But we can probably do better indeed through the dbviewer.

> In particular, do we plan to maintain the Oracle database behind the dbviewer? That would also require us to have a relatively stable data model. My plan up till now was to stop using that dbviewer, [...]

I also have control over the MICADO Oracle database.

> We should formulate a plan about what we plan to support, and for who.

For now, my goal is to support the DFS. I'll leave it to you to decide when/whether we want to support the rest of the consortium.

> What we perhaps can do is fully automate the generation of whatever data we want in this archive. then we can recreate all the data each time we merge new code into develop. Then we simply throw the [...]

Yes that was what I'm hoping to achieve.

That would mean that we need to ensure that MICADO-WISE can ask ScopeSIM simulate everything that the science team wants to have. But maybe we can take some shortcuts.
 
> And for now we would probably still need to use an Oracle database, since not everything the dbviewer does is possible with PostgreSQL yet.

I'm not sure what the status is of the DB viewer w.r.t to PostgreSQL. It was my impression that although not everything works, that it should work well enough for most of our current needs. I would prefer to not support the Oracle database if at all feasible.

hugobuddel, 2023-03-23:
I'm closing this, as it is not really a ScopeSim issue. We have a MicadoWISE archive, so that's good.

closed 2023-03-23 by hugobuddel