AstarVienna/ScopeSim #25: Installation / build fails because scopesim/version.py is missing

hugobuddel, 2020-06-17:
Installing or building ScopeSIM fails because `scopesim/version.py` is missing: https://travis-ci.org/github/astronomyk/ScopeSim/jobs/698989265
```
$ pip install .
Processing /home/travis/build/astronomyk/ScopeSim
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-j2npccl5/setup.py", line 20, in <module>
        with open('scopesim/version.py') as f:
    FileNotFoundError: [Errno 2] No such file or directory: 'scopesim/version.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-j2npccl5/
The command "pip install ." failed and exited with 1 during .
```

astronomyk, 2020-06-17:
Thanks for makeing the issue. I read the error yesterday, but didn't react because of other priorities. Now it should be fixed =)
THanks for having my back.
I'll close this now

closed 2020-06-17 by astronomyk

astronomyk, 2020-06-17:
Thats wierd because I have version.py on my local machine, but also found
version.py in my .gitignore.
Let me check github

-------------------------------------
Kieran Leschinski
Department of Astrophysics
University of Vienna

hugobuddel, 2020-06-17:
It works again, thanks. I created the issue because I was not sure you noticed it yesterday. 

@gijsverdoeskleijn we can use the master branch again to install ScopeSIM

astronomyk, 2020-06-17:
Now its broken again :/

-------------------------------------
Kieran Leschinski
Department of Astrophysics
University of Vienna

hugobuddel, 2020-06-17:
The build logs seems to indicate that everything is fine for Python 3.7 and 3.8, and since we only use those (f-strings), everything should be fine: https://travis-ci.org/github/astronomyk/ScopeSim/builds/699297462

However, it would be nice if you could try to keep master working; that is, only merge things into master if the CI succeeds. For some reason master seems to break exactly when @gijsvereskleijn decides to do something :tired_face:

astronomyk, 2020-06-17:
FWIW This is an issue with the older version of synphot+astropy. As long as
you're using the synphot 2+ and astropy 3+ all will be fine :)