AstarVienna/ScopeSim #26: 5-line script loading_packages.rst does not work

hugobuddel, 2020-06-22:
https://github.com/astronomyk/ScopeSim/blob/A_5_liners/docs/source/use_examples/5-liners/loading_packages.rst
does not seem to work:
```
(base) [root@0fab83c32ed3 /]$ python
Python 3.8.3 | packaged by conda-forge | (default, Jun  1 2020, 17:43:00)
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import scopesim
>>> scopesim.rc.__config__["!SIM.file.local_packages_path"] = "./temp/"
>>> pkg_names = ["locations/Paranal", "telescopes/VLT", "instruments/HAWKI"]
>>> scopesim.server.download_package(pkg_names)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/conda/lib/python3.8/site-packages/scopesim/server/database.py", line 166, in download_package
    save_path = [download_package(pkg, save_dir, url) for pkg in pkg_path]
  File "/opt/conda/lib/python3.8/site-packages/scopesim/server/database.py", line 166, in <listcomp>
    save_path = [download_package(pkg, save_dir, url) for pkg in pkg_path]
  File "/opt/conda/lib/python3.8/site-packages/scopesim/server/database.py", line 183, in download_package
    file_path = shutil.copy2(cache_path, save_path)
  File "/opt/conda/lib/python3.8/shutil.py", line 432, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/opt/conda/lib/python3.8/shutil.py", line 261, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: './temp/Paranal.zip'
>>>
```

hugobuddel, 2020-06-22:
It works after creating the temp directory:
```
>>> import os
>>> os.mkdir("./temp")
>>> scopesim.server.download_package(pkg_names)
Downloading https://www.univie.ac.at/simcado/InstPkgSvr/telescopes/VLT.zip
|============| 2.6M/2.6M (100.00%)         7s
Downloading https://www.univie.ac.at/simcado/InstPkgSvr/instruments/HAWKI.zip
|============|  25k/ 25k (100.00%)         0s
['/temp/Paranal.zip', '/temp/VLT.zip', '/temp/HAWKI.zip']
>>>
```

astronomyk, 2020-06-22:
Awesome! That'll explain why readthedocs is having trouble too.

hugobuddel, 2020-06-22:
You could perhaps use the `tempfile` library to create the directory.

hugobuddel, 2023-03-23:
Closing, as downloading the packages will now create the directory if it doesn't exist.

closed 2023-03-23 by hugobuddel