AstarVienna/ScopeSim #30: Calibration data should have persistent structure

hugobuddel, 2020-06-23:
The image artifacts that are introduced by the instrument have a relatively stable structure, this should be represented in the simulations.

E.g. the quantum efficiency varies per pixel, but is relatively stable over time. Therefore it is possible to recover the quantum efficiency by observing many flat frames and averaging them: the master flat will show this structure. Similarly for the dark and illumination correction.

However, this structure is random in the current simulations. E.g. simulating a raw dark frame can be done as described in [test_simplecado.py](https://github.com/astronomyk/ScopeSim/blob/master/scopesim/tests/tests_integrations/test_simplecado.py). However, this will always result in a dark with an entirely different structure.

Two possible ways to solve this (that can be combined):
* allow the random seed to be specified for the effect (dark current / quantum efficiency),
* allow the structure to be given as a FITS file.

Being able to set the random seed would also generalize to other effects. ESO seems to already have some header keywords for this, @hugobuddel will look those up.

astronomyk, 2020-06-23:
Just a quick comment on the random seed - this is already possible. Just use the following config keyword:

    rc.__config__["!SIM.random.seed"] = 42

Also possible to directly set this in each Effect discription in the yaml files:

For example in ``MICADO_H4RG.yaml`` we could add:

    -   name: shot_noise
        class: ShotNoise
        kwargs:
            - random_seed: 42

The same works for the ReadoutNoise effect

This would be a nice quick 5-liner for the documentation, as the ``!SIM.random.seed`` is called where-ever a random number or sequence is needed in scopesim

hugobuddel, 2020-06-24:
For the 5LS see #34 (just linking the issues).

However, this would not really solve the issue. The pattern should be generated with a fixed seed, but on top of that there should be shot noise with a different seed for each exposure. (The noise can just be on the total signal I think, but maybe there should be specific noise for different signal sources.)

There is no pattern generation in either `DarkCurrent` or `QuantumEfficiencyCurve`. So I think we should either

* Add this (random) pattern generation to `DarkCurrent` or `QuantumEfficiencyCurve`.
* Add support for using a FITS file for the `DarkCurrent` or `QuantumEfficiencyCurve` (which for simplicity might be created through simple Poisson noise like now).
* Best: do both.