AstarVienna/ScopeSim #38: Effects objects To-Do list

astronomyk, 2020-07-22:
# Overview of Effects that need to be added to ScopeSim

A seperate issure should be created for each of these objects and linked to this post

## Observations

- [ ] Dithering
- [ ] Field Rotations
- [ ] Pupil rotation PSF

## Optics
- [ ] Distortion / Projections
- [ ] Spatially dependent background emission

## Electronics
- [ ] Cosmics for CCDs
- [ ] Cross-talk
- [ ] Persistence
- [ ] Hot/dead pixels
- [ ] Averaged exposure readout

hugobuddel, 2020-07-22:
The priorities for the data processing of MICADO is on the dithering so we can simulate a dithered observation. I'm planning to simply move the source objects for now.

hugobuddel, 2023-03-22:
Hi @joao-aveiro , thanks for your interest. I believe most of these other Effects are not yet implemented.

You started using ScopeSim at a good time (assuming you started using it recently). The last ~year there has been (too) little development; mainly because projects, and therefor contracts, were ending. However, now most (not all..) ScopeSim developers are employed again in positions that require ScopeSim, so we're picking up steam again.

I've taken over the role of maintainer of ScopeSim (and assorted packages). My goal is to make it a robust open source project, so it will be easier to contribute (like you did). So I'm fixing immediate bugs, updating the documentation, triaging the issues, etc. So your thoughts on what is important are appreciated.

The core ScopeSim team will probably focus on adding features to the core of ScopeSim. For example being able to simulate full templates, random number stuff, time-dependence (e.g. for persistence), etc. My hope is that we can make contributing to ScopeSim pleasant enough that we can attract interested people to assist with adding the effects they are passionate about.