AstarVienna/ScopeSim #42: ScopeSim September Workload

astronomyk, 2020-08-24:
Coding
------
- [ ] (MV, OC) finish writing report() functions for all Effects
- [ ] (OC) Detector readout FITS header
- [ ] (KL) write template for UserCommands report functionality
    - [ ] make sure to load all yamls
    - [ ] generate an OpticalElement report for each yaml
- [ ] (KL) finish up new FOVManager branch
- [ ] (KL) stop SkyCalc TERCurvce from requiring an internet connection on startup

Packages
--------
- [ ] (KL) finish MICADO sci-team package
    - [ ] move MICADO_sci.yaml doubled functionality to MICADO package
- [ ] (??) synchronise header info in all files with online docs interface
- [ ] (OC) debug METIS package

Testing
-------
- [ ] (MV, OC) HAWKI limiting magnitudes test
- [ ] (MV, OC) HAWKI flux test. Test against any raw archive data. Best would be data taken for SimCADO

FDR Documentation
-----------------
- [ ] (MV, OC, KL) Use Case docs (ScopeSim Manual)
    - [ ] Collect the main MICADO science cases and plan a series of use case examples
    - [ ] Possible tests could also be written "properly" so that we can include them in the docs
- [ ] (MV, OC, KL)ScopeSim environment Overview
    - [ ] (KL) Introduction
    - [ ] (KL) ScopeSim
    - [ ] (MV) ScopeSim_templates
    - [ ] (MV) SpeXtra
    - [ ] (OC) IRDB
    - [ ] (KL) AnisoCADO
    - [ ] (KL) SkyCalc_ipy
- [ ] (Auto) IRDB MICADO summary
    - [ ] (KL) Write introduction for the doc

Docstrings 
----------
- [ ] (MV, OC) Effect Docstrings
    - [ ] Write a description of each Effect object
    - [ ] Where possible write an example for loading from a file and loading from direct python input
- [ ] (KL) ScopeSim Core docstrings

teutoburg, 2023-09-29:
_[looking through stale issues]_

@astronomyk 
This is now one year old and hasn't seen any updates in more than 6 months. Not all people mentioned here are still around. I suggest to close the issue and move all still relevant TODOs to somewhere else.

teutoburg, 2024-10-03:
> _[looking through stale issues]_
> 
> [@astronomyk](https://github.com/astronomyk) This is now one year old and hasn't seen any updates in more than 6 months. Not all people mentioned here are still around. I suggest to close the issue and move all still relevant TODOs to somewhere else.

[image]
@astronomyk