AstarVienna/ScopeSim #116: [Improvement suggestion] Extra FITS header keywords to be added to the pipeline FITS files

astronomyk, 2022-03-24:
A suggestion for how to add extra FITS header keywords to the pipeline FITS files
---------------------------------------------------------------------------------

This suggestion is ONLY for keywords that should be MANUALLY ADDED (in an "automated" fashion) to the
headers after a simulation is read-out by the detector.

The idea here is that we can add a single ``.yaml`` file to the IRDB directory that contains all the needed keywords for ESO deliveries.

Simulation parameters (Effect kwargs values, etc) will be added automatically
by ScopeSim in a different function, but following this format.

This file should be split into different HIERARCH lists:

- HIERARCH ESO
 Only for ESO specific keywords
- HIERARCH SIM
 For ScopeSim specific keywords, like simulation parameters
- HIERARCH MIC
 For MICADO specific keywords, (unsure what these would be yet)

More HIERARCH style keywords can also be added as needed for other use-cases, e.g. MAORY

Document format
---------------
This document is a yaml document.
Hence all new keywords should be specified in the form of yaml nested
dictionaries.
As each ``astropy.HDUList`` contains one or more extensions, the initial level is
reserved for a list of keyword groups.
For example::

    - ext_type: PrimaryHDU
      keywords:
        HIERARCH:
          ESO:
            ATM:
              TEMPERAT: -5
              PWV: 1.0

    - ext_number: [1, 2]
      keywords:
        HIERARCH:
          ESO:
            DET:
              DIT: [5, '[s] exposure length']   # example of adding a comment
              NDIT: 1

The keywords can be added to one or more extensions, by using one of the
following ``ext_`` qualifiers: ``ext_name``, ``ext_number``, or ``ext_type``

Each of these ``ext_`` qualifiers can be a ``str`` or a ``list``.
For a list, ScopeSim will add the keywords to all extensions matching the
specified type/name/number

The above example will result in the following keyword added to:

- PrimaryHDU (ext 0):

    header['HIERARCH ESO ATM TEMPERAT'] = -5

- Extensions 1 and 2 (regardless of type):

    header['HIERARCH ESO DET DIT'] = (5, '[s] exposure length')

Resolved and un-resolved keywords
---------------------------------
ScopeSim uses bang-strings to resolve global parameters.
E.g: ``from_currsys('!ATMO.temperature')`` will resolve to a float
These bang-strings will be resolved automatically in the ``keywords`` dictionary
section.

If the keywords bang-string should instead remain unresolved and the string
added verbatim to the header, we use the ``unresolved_keywords`` dictionary
section.

Additionally, new functionality will be added to ScopeSim to resolve the
kwargs/meta parameters of Effect objects.
The format for this will be to use a new type: the hash-string.
This will have this format::

    #<optical_element_name>.<effect_name>.<kwarg_name>

For example, the temperature of the MICADO detector array can be accessed by::

    '#MICADO_DET.full_detector_array.temperature'

In the context of the yaml file this would look like

    - ext_type: PrimaryHDU
      keywords:
        HIERARCH:
          ESO:
            DET
              TEMPERAT: '#MICADO_DET.full_detector_array.temperature'    # resolves to a float
      unresolved_keywords:
        HIERARCH:
          ESO:
            DET
              TEMPERAT: '#MICADO_DET.full_detector_array.temperature'    # stays as this string

Obviously some thought needs to be put into how exactly we list the simulation
parameters in a coherent manner.
But this is 'Zukunftsmusik'.
For now we really just want an interface that can add the ESO header keywords,
which can also be expanded in the future for our own purposes.

Below is an example of some extra keywords for MICADO headers

    - ext_type: PrimaryHDU
      keywords:
        HIERARCH:
          ESO:
            ATM:
              TEMPERAT: "!ATMO.temperature"   # will be resolved via from_currsys
              PWV: "!ATMO.pwv"
              SEEING: 1.2
            DAR:
              VALUE: "#MICADO_DET.dark_current.value"   # will be resolved via effects
            DPR:
              TYPE: "some_type"
          SIM:
            random_simulation_keyword: some_value
          MIC:
            micado_specific: ["keyword", "keyword comment"]

      unresolved_keywords:
        HIERARCH:
          ESO:
            ATM:
              TEMPERAT: "!ATMO.temperature"   # will be left as a string

    - ext_type: ImageHDU
      keywords:
        HIERARCH:
          SIM:
            hello: world
            hallo: welt
            grias_di: woed
            zdrasviute: mir
            salud: el mundo

hugobuddel, 2022-03-24:
Thanks @astronomyk , a good summary of our discussion yesterday!

What seems missing, is that several keywords will depend on the 'template' that is being 'observed', see the MICADO Template Manual. We had some discussion whether it also changes which keywords are included, or that all raw data has the same keywords, just some with an empty/None/default value. I don't think that is maintainable, but perhaps we can assume that for now all data has the same keywords. However, the values of some keywords would change depending on 'template'.

My proposal is to keep this file above as you just described it; a single file that defines the keywords of all raw data. Then we make a new group of 'object' (next to ATMO/TEL/INS/DET), called 'template' (TPL) or so, with several yaml files for each project. These Template objects would have no effects, but only properties, e.g. with DPR.TYPE values. The only goal of those properties is to be used in this new yaml file you described here.

hugobuddel, 2022-03-24:
A note about toekomstmuziek: besides the "HIERARCH ESO" keywords, I want to, somehow, get header keywords that allow these files to be ingested into the MicadoWISE archive. It seems that I can use this proposed file to achieve that already now.

I would prefer to reuse as much as the HIERARCH ESO keywords for that, and otherwise the HIERARCH SIM keywords, but failing that, this structure makes it trivial to add a HIERARCH MCD or MWS whatever. Then we can get the best of everything I think. In particular, the MicadoWISE keywords solve the reproducibility, so they could directly be used as HIERARCH SIM keywords. However, that would require consensus with the other instrument teams, and that would take more time. So I can use the HIERARCH MWS ones for now, and over time harmonize those with the HIERARCH SIM ones (in either direction).

hugobuddel, 2022-03-24:
One other thing that might be missing: some of these header keywords could perhaps be values that are calculated / derived inside ScopeSIM. As in, that the values are not directly available in the input yaml files.

I cannot give any actual examples now, so maybe there aren't any currently. An example could be the time that the ADC rotates to a new position.

I'd say that one way to resolve this is to add all of those to the input yaml files anyway (and thus end up in the rc dictionary), with values that are left empty, and are filled up during the simulation. Then they can be accessed with the same mechanism as above.

That solution would be especially useful for values that normally *should* be derived, but that we might want to hardcode occasionally anyway. Again, the ADC rotation would be a good example. In a way, the template start is such a value.

But if we don't have such header keywords now, then we shouldn't bother now.

hugobuddel, 2022-03-24:
> These Template objects would have no effects, but only properties, e.g. with DPR.TYPE values. The only goal of those properties is to be used in this new yaml file you described here.

We could perhaps, in the future, use these template objects to also do other things than defining header keywords. For example dither patters can be described through templates. And certain effect configurations are enforced (or forbidden) by specific templates. Etc. It is not necessary to add all that functionality now, but the possibility is an argument for introducing these template objects.

hugobuddel, 2022-03-24:
> My proposal is to keep this file above as you just described it; a single file that defines the keywords of all raw data. Then we make a new group of 'object' (next to ATMO/TEL/INS/DET), called 'template' (TPL) or so, with several yaml files for each project.

Another option is to not make a new kind of yaml file, at least not now, but put all those values in the observation dictionary and refer to that. That observation dictionary already exists, so this removes the need to add another extra new thing. However, this solution would require the user to put in all these values all the time, which is only a short term solution (which might be fine for now).

astronomyk, 2022-03-29:
@hugobuddel 

Here's my initial attempt at a fits header dict flattening Effect. 

https://github.com/AstarVienna/ScopeSim/compare/dev_master...kl/fits_effect?expand=1

It works on the structure defined here:

https://github.com/AstarVienna/irdb/blob/dev_micado/MICADO/FITS_extra_keywords.yaml

I have tests for the support functions, but not for the Effect itself yet. They will come tomorrow. In the mean time, you can see where this is going, so feel free to start adding whatever you think is relevant to the FITS_extra_keywords.yaml file

hugobuddel, 2022-03-30:
Thanks @astronomyk ! I'll add some header keywords tomorrow or Friday and see how things go.

I think it is good to make it just another effect. That way it would be easy to specify a different yaml file, or multiple, and also to extend this construct to proper instrument templates somehow.

closed 2022-04-11 by astronomyk