AstarVienna/ScopeSim #130: suggestion: Provide default copy of all instrument configs with ScopeSim installation

krachyon, 2022-04-08:
I see no particular reason for ScopeSim to require you to manually download the correct set of config files and write them into the working directory of the interpreter. It would be more intuitive to my mind the following way:

```python
# pre-defined constant for the instrument, loads the default config
micado = scopesim.OpticalTrain(scopesim.Instruments.MICADO)
# I have a directory of config files that I edited, please build an OpticalTrain from that
my_micado = scopesim.OpticalTrain(pathlib.Path('/my/config/files'))
# or even more explicit
my_micado = scopesim.OpticalTrain.from_path(pathlib.Path('/my/config/files'))
```

In any case the "use whatever is in `os.getcwd()` " felt like a footgun and I decided to wrap all calls to ScopeSim in a decorator so that it always used a common directory (and I didn't have to call download so many times)

hugobuddel, 2022-04-08:
Yes, because there is also a problem that it is not clear which version of the instrument packages are used by ScopeSIM.

Al instrument information comes from the IRDB: https://github.com/AstarVienna/irdb . The easiest way to get that nicely into ScopeSIM might be to simply make the entire IRDB a python package (or multiple packages) and make it a dependency of ScopeSIM. Then one can simply `pip install -e .` when one wants to use a local version.

Your `scopesim.Instruments.MICADO` would then be something like `irdb.MICADO`. It would perhaps be nice if these would be actual Python objects representing this information.

teutoburg, 2025-08-31:
An alternative to this has been implemented (a while ago) in the form of the "scopesimple" Interface, so this is no longer relevant.

closed 2025-08-31 by teutoburg

hugobuddel, 2025-08-31:
don't we still want to make pip packages of the irdb? that's still on my wishlist somewhere

teutoburg, 2025-08-31:
Yes, and in my mind closing this issue doesn't mean we don't. But that's a somewhat different discussion, and one that should IMO be handled over in the IRDB. But for now, there's a good enough alternative in the `Simulation` class, that does solve the original complaint in this issue, which is having to _manually_ download IRDB packages.

teutoburg, 2025-08-31:
In hindsight, I could have instead closed this as "completed", but it felt more like "resolved in another way". But whatever.