AstarVienna/ScopeSim #157: SCAO vs MCAO, PSF and fluxes behaviour in extended and star mode

carmeloarci, 2022-07-15:
MCAO vs SCAO of MICADO
Using as source a star, both configurations use the same PSF (SCAO)
Sky Background in SCAO is about half (60% of )  w.r.t. the MCAO 
The same star has a peak in MCAO that is 2% of the SCAO (Ks) 
Using as source an image (galaxy), the PSF are different and fluxes are similar
[AWG 3-06-2022.pptx](https://github.com/AstarVienna/ScopeSim/files/9119461/AWG.3-06-2022.pptx)
[Testing SCAO and MCAO.ipynb.zip](https://github.com/AstarVienna/ScopeSim/files/9119476/Testing.SCAO.and.MCAO.ipynb.zip)

carmeloarci, 2022-07-15:
@astronomyk is the assignee

hugobuddel, 2023-03-21:
Thanks @carmeloarci . I've assigned @astronomyk Kieran, because this would need Kierans input to resolve.

Below are my thoughts, summarizing / rephrasing this issue. It contains a few questions that I perhaps should discuss with @astronomyk first, to get myself up to speed, to make a discussion with you more productive. Nevertheless, I prefer to do things in the open, so you are welcome to chip in as well @carmeloarci .

I'll classify this as a bug, since it seems that something is indeed not working as intended.

### PSF
Using MCAO (with a star as source) should use the MCAO PSF, but instead it uses the SCAO PSF, right?

I'm not sure how it is determined which PSF to use, or which PSF is actually in use. I would expect ScopeSIM can choose between these two PSFs:
  * https://github.com/AstarVienna/irdb/blob/dev_master/MAORY/PSF_MCAO_ConstPSF_40_18_6.fits
  * https://github.com/AstarVienna/irdb/blob/dev_master/MICADO/PSF_SCAO_ConstPSF_0_5off.fits

To me those two PSFs look very similar, and I personally would probably not be able to tell from the science data which one was used. Are these two PSFs correct? What is the difference between them?

### Sky Background SCAO vs MCAO
Indeed, it seems that the SCAO background is about 60% of the MCAO background for the star. However, they seem equivalent for the galaxy.

We would expect the background flux to be equivalent irrespective of whether SCAO or MCAO is used?

### Peak of star

You state that the same star has a peak in MCAO that is 2% of the SCAO (Ks). I've not been able to verify this; the stars look kinda equivalent at a first glance. But I have not created proper histograms and such. Is this still a problem?

### Galaxy
You state that "Using as source an image (galaxy), the PSF are different and fluxes are similar". I've verified this, but it is not clear whether you consider this good or bad. It seems to me that this is the desired behavior, is that correct?

I understand that it is bad that the star behaves differently from a galaxy, but I'm not sure what the correct behavior is.

carmeloarci, 2023-03-21:
Hi @hugobuddel 
let me share my original results here 
https://drive.google.com/drive/folders/1k54puxUuVOdd5qRdFPGTq5gv9zuo063O?usp=share_link
the pdf and the notebook are in the shared folder (however these two were already attached).

-- **SKY BACKGROUND** --
>We would expect the background flux to be equivalent irrespective of whether SCAO or MCAO is used? 
Almost identical sky level, somewhat higher for mcao. 
I tuned the original notebook by adopting the same filter and DIT to both gal and star cases. Looking to the results I don't see problems. Using this modified notebook, I tested that the both extended source and star has the same background. Moreover I also checked the difference in the background between MCAO and SCAO: at the time of the opening of the issue, in the star case, the background for MCAO was smaller than in SCAO, and the opposite in the galaxy case (was in the H band). Now I find no difference in galaxy vs star cases.  And the difference between MCAO and SCAO fou
[Testing SCAO and MCAO.pdf](https://github.com/AstarVienna/ScopeSim/files/11034770/Testing.SCAO.and.MCAO.pdf)
nd could match the difference in the emissivity/reflectivity of the two configurations. 
SOLVED? WE SHOULD CHECK IN THE EMISSION/REFLECTIVITY files for MCAO and SCAO.

-- **PEAK of star** ---
> the stars look kinda equivalent at a first glance. But I have not created proper histograms and such. Is this still a problem?
Not. Actually, the fluxes, between SCAO and MCAO cases nicely match expectations. And I found back the PSF you were pointing:
    https://github.com/AstarVienna/irdb/blob/dev_master/MAORY/PSF_MCAO_ConstPSF_40_18_6.fits
    https://github.com/AstarVienna/irdb/blob/dev_master/MICADO/PSF_SCAO_ConstPSF_0_5off.fits
Also using the modified notebook I don't see problems.
SOLVED

-- **Galaxy** --
Regarding the The star images match (SCAO/MCAO), the galaxy images  match (SCAO/MCAO). The star and galaxy images may match according to the definition of "magnitude" in the extended source case:
- if it is an integrated magnitude they don't (too much flux in the galaxy case) 
- if it a kind of magnitude per square arcsec or some other, it may be ok. 
If it is the second case it's OK.
[Testing SCAO and MCAO mod.pdf](https://github.com/AstarVienna/ScopeSim/files/11034787/Testing.SCAO.and.MCAO.mod.pdf)

hugobuddel, 2023-03-24:
Thanks @carmeloarci. These kind of consistency checks are important to get right, so I'll go through your scenario's in more detail. At least we should understand any discrepancies (and explain them properly).