AstarVienna/ScopeSim #171: ScopeSIM error on FITS header

gijsverdoeskleijn, 2022-11-17:
Gijs-to-self: [mailthread](https://mail.google.com/mail/u/0/?tab=rm#search/from%3Avriend+slack/FMfcgzGqRZXPBJVtbpLmdKNKDGkFsQSG)

oczoske, 2022-12-13:
I think you'd better ask these questions on slack... I don't know which fits error you're referring to, but one of the following might be what you're looking for:
```
micado = sim.OpticalTrain(cmds)
micado['source_fits_keywords'].include = False
micado['extra_fits_keywords'].include = False
micado['extra_fits_keywords_ro'].include = False
```

gijsverdoeskleijn, 2022-12-13:
> ....one of the following might be what you're looking for

Thanks @oczoske , cc @astronomyk : turning those off indeed bypassed the piece of code that creates the error.

hugobuddel, 2023-03-21:
This should be a bit better now, but not sure it fully works. What we should do:

* Create a bit more complicated testcase than what we have now, to ensure these headers work properly. Perhaps based on Gijs' notebook.
* Ensure there is a proper fallback in case these headers are causing problems, because this should not be a problem that breaks the simulator.
* Properly document what to expect from these effects.

hugobuddel, 2023-03-21:
I've marked this as duplicate, because I think it is a duplicate of #136