AstarVienna/ScopeSim #174: IMG_4mas mode shows weird discontinuity with cross patterns

yixiancao, 2023-01-13:
In the 4mas mode, all images (except DARK) show weird discontinuity with cross patterns.  
Below is the Flat image, and these patterns persist for all the other illuminated images.  
[image]
Another example is a zoomed in image of a stellar field in seeing limited case.  The image is discontinuous at the crosses. 
[image]

The full simulation data set is at https://nextcloud.mpe.mpg.de/nextcloud/s/PJ4xd47Tb24gLR9, including the notebooks generating it.

hugobuddel, 2023-03-21:
This is probably the same bug that causes the discontinuity that can be seen in https://github.com/AstarVienna/irdb/blob/dev_master/MICADO/docs/example_notebooks/7_dithered_cluster.ipynb

This might be related to the off-by-one error that also caused #111 and that can also be seen in https://github.com/AstarVienna/irdb/blob/dev_master/MICADO/docs/example_notebooks/2_scopesim_SCAO_1.5mas_astrometry.ipynb and 

Thanks for the notebooks. Perhaps a good first step is to try to see whether we can reproduce the issue with earlier ScopeSim versions or whether the problem has always been there.

teutoburg, 2024-04-17:
Does this still occur as of v0.8.0?