AstarVienna/ScopeSim #191: Need for nightly test runs

hugobuddel, 2023-03-21:
Nightly test-runs are needed, because dependencies are not pinned. Therefore a new version of a dependency, e.g. of astropy, might cause bugs. See for example

https://github.com/AstarVienna/irdb/actions/runs/4478137091/jobs/7870528319

We need this in all A*V projects

hugobuddel, 2023-03-21:
And the tests should extend to Python 3.11

hugobuddel, 2023-03-21:
This needs to be fixed as well:

```
Node.js 12 actions are deprecated. Please update the following actions to use Node.js 16: actions/checkout@v2, actions/setup-python@v2. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/.
```

teutoburg, 2023-11-03:
Is this implemented already? AFAIK we're running nightly tests for a while now...