AstarVienna/ScopeSim #199: Ensure minimum versions for dependencies are correct

hugobuddel, 2023-04-04:
ScopeSim only works with recent enough versions of e.g. speXtra, but this is not yet reflected in `setup.py`. So `pip upgrade ScopeSim` will not update  the dependencies, and therefore things won't work until one also runs e.g. `pip upgrade spextra`. See https://github.com/AstarVienna/ScopeSim_Templates/issues/49#issuecomment-1494233399 . When the versions of the dependencies are correct in `setup.py`, then just upgrading ScopeSim is enough. A similar problem probably also occurs with other dependencies, and also the other projects like ScopeSim_Templates.

Things that need to happen:

* The minimum versions need to be correct in `setup.py`, and also in the `requirements.*`.
* There needs to be some kind of test that automatically tries to install ScopeSim with these lowest version dependencies to see whether they work.
* This needs to happen in all projects, but maybe ScopeSim would be sufficient, because that is the main point of entry.
*

teutoburg, 2023-11-03:
I think this should be solved by our current use of `pyproject.toml`, can we close this?

hugobuddel, 2023-11-03:
Yeah. Works well enough at least

closed 2023-11-03 by hugobuddel