AstarVienna/ScopeSim #207: Should ScopeSim_Templates be a dependency of ScopeSim?

hugobuddel, 2023-05-06:
Currently ScopeSim is a dependency of ScopeSim_Templates. However, it might be better if it is the other way around, that is, that ScopeSim_Templates is a dependency of ScopeSim. In the latter case, people who install or update ScopeSim, will automatically get the right ScopeSim_Templates.

hugobuddel, 2023-05-06:
It seems perfectly possible to have both packages be a dependency of the other, so maybe that would be the solution?

JuHeart, 2023-06-01:
something like this, but the default user definitely would prefer a simple, one line solution to update everything related to ScopeSim (including IRDB if IRDB is already installed...)

teutoburg, 2023-11-03:
This is part of the "bigger picture" of #293 ...