AstarVienna/ScopeSim #211: NCPA and PSF affect MICADO spectroscopy with homogeneously filled slit in a weird way

oczoske, 2023-05-25:
A homogeneous extended source (e.g. sky emission or calibration lamp) should not be affected by PSF convolution. This is also true for spectroscopy where the slit is homogeneously filled. The figure shows a section of MICADO long-slit spectroscopy with a 20mas wide slit (mode `SPEC_3000x20`) in the `Spec_HK` filter. The source is `scopesim_templates.micado.flatlamp`, where the spectrum has been replaced by a presmoothed spectrum with equal-strength lines at fixed wavelength separations. The irdb setup for this mode has two `micado_ncpas_psf` (subclass of `AnalyticalPSF`) and `relay_psf` (`FieldConstantPSF`). 
The top panel has both switched off, the second panel has `ncpas_psf` turned on, the third panel has `relay_psf` turned on, and the bottom panel has both. Without any PSF a line has a length of about 500 pixels (2000 mas), i.e. shorter than the slit length. The NPCA causes the appearance of the outlined rectangle (very weird!) that extends the trace to the left, giving it the full length of 750 pixels (3000 mas). The relay PSF alone fills this rectangle. Adding in the ncpa causes a slight widening of the lines.   
[image]

oczoske, 2023-06-20:
[micado_vs_psf_ncpa_ycut.pdf](https://github.com/AstarVienna/ScopeSim/files/11801929/micado_vs_psf_ncpa_ycut.pdf)
Full cuts through a trace (ORDER_4_1, y-offsets applied to separate the curves). Simulations with effect `relay_psf` cover the full slit length of 3000 arcsec (750 pixels), those without are only 500 pixels wide. The effect is asymmetric, as the traces are only broadened on one side. ncpa raises the floor very slightly and adds the edge seen above.

oczoske, 2023-06-28:
Here's what I think is going on: The source `scopesim_templates.micado.flatlamp.flatlamp()` creates a 16x16 ImageHDU with pixel scale 1.024 arcsec, much larger than the pixel scale of MICADO. `fov_utils.extract_area_from_imagehdu()`  cuts out a section that is supposed to cover the field of view, which is defined by the slit aperture in this case. This results in a curious 2x0 ImageHDU (effectively 2x1) through `data = imagehdu.data[y0p:y1p, x0p:x1p]`.  This is not large quite large enough in the spatial direction. Changing the line to `data = imagehdu.data[y0p:y1p+1, x0p:x1p+1]` works for MICADO and results in the correct trace width for all four cases. 

For METIS, this change  was not necessary, the traces had the correct width all along. 

My conclusion is that `extract_area_from_imagehdu()` (and all its siblings...) needs to check the pixel edges instead of just the pixel centres to ensure that the entire fov volume is covered by the ImageHDU.

closed 2023-06-29 by hugobuddel

hugobuddel, 2023-06-29:
I would not expect that my choice of 16x16 would actually be useful in highlighting a problem! Good work @oczoske

PR #238 (2023-06-29): extract from imagehdu so that full fov is covered
This fixes #211, compare 
[micado_vs_psf_ncpa_ycut_correct.pdf](https://github.com/AstarVienna/ScopeSim/files/11896228/micado_vs_psf_ncpa_ycut_correct.pdf) to the figure linked there. Instead of looking at the pixel edges I replaced `round()` by the more appropriate `floor()` and `ceil()` functions and made sure that the proper pixel range is included.