AstarVienna/ScopeSim #212: MICADO spectroscopy: line widths are too large

oczoske, 2023-05-25:
For a homogeneously filled slit the line shape in MICADO long-slit spectroscopy should be dominated by the slit profile; an unresolved (monochromatic) emission line should produce an image of the slit at some position in the detector plane (determined by the spectral mapping). The figure shows simulations of a regular spaced line lamp with modes `SPEC_3000x20` and `SPEC_3000x50`, respectively (cf. issue #211 ). The lines have widths of roughly 20 and 50 *pixels*, whereas they should be around 20 and 50 *mas*. (these simulations were done without any PSF convolution).
[image]

oczoske, 2023-05-26:
This issue may be due to the fact that those simulations used a `detector_window` of 4096x4096 pixels rather than the `full_detector_arrray`. The dispersion `dlam_per_pix` used to build the `XiLamImage` is computed in `spectral_trace_list_utils.py` (around l.215) as the average dispersion of the range `wave_min` to `wave_max` over `sub_naxis2` pixels. This is set to 4096, whereas the fov cube has 21201 pixels for the wavelength range. Full-array simulations may come out right; still the issue needs to be fixed (I can't do full-array simulations on my laptop) and the computation of the dispersion refined.

oczoske, 2023-05-29:
The figure shows the dispersion `dlam_per_pix` as a function of wavelength for the MICADO traces. This is in the y-direction only, computed as `xy2lam.gradient()[1](x, y) * pixsize` along the centre of the slit (more precisely `xi = 0`). The variation with wavelength is small (for a given trace), which makes use of an average `dlam_per_pix`  acceptable, as the variation has a negligible effect on the line width (the mapping uses the nonlinear transformation). However, inclusion of the full wavelength-dependent dispersion is easily possible.
[image]

JuHeart, 2023-05-30:
Hi Oliver, 
we tried to verify that with full view mode (all detectors), it works better. It is indeed closer to the expectation, but we identified still a strange mismatch (or misunderstanding on our side): if we go into spectral_trace_list_utils.py and print, then the dlam_per_pix seems still too high with respect to the dispersion plots just above. For instance in K-band, we get 4.87e-5 µm/pix instead of approx 4.3e-5µm/pix from the order_3_1 plot above. Consistently, in our filled 48mas slit spectrum the spectral resolution seems R~4000k instead of the R~3300 report in OCD table11. With a disperion as plotted above (4.3e-5µm/pix), it is much closer to the nominal 3300. So the dlam_per_pix might still be off in ScopeSim. Could you check this? 

These are the numbers spectral_trace_list_utils.py calculates around l.215
> dlam per pix:4.8728309959591175e-05
> wave min:1.875
> wave max:2.49
> sub_naxis1:1875
> sub_naxis2:12621
> 
> dlam per pix:3.4234961647153826e-05
> wave min:1.446
> wave max:1.87
> sub_naxis1:947
> sub_naxis2:12385
> 
> dlam per pix:4.1666666666666706e-05
> wave min:1.446
> wave max:1.5
> sub_naxis1:893
> sub_naxis2:1296
> ```

JuHeart, 2023-05-30:
then, if we use the option detector window (so only a part of the HK filter) and the dispersion number (which should be stable) change drastically, so there seems something inconsistent in defining the wave_min wave_max, or the sub_naxis values...

JuHeart, 2023-05-30:
yes, indeed I think I mixed up 'closing' with 'commenting'... thanks for noting, and for working on a fix