AstarVienna/ScopeSim #219: Explicitly specify encoding

teutoburg, 2023-06-12:
## TLDR

Add `encoding="utf-8"` kwarg to all `open()` calls to prevent issues with encoding.

## The issue

Encoding is not trivial when opening a file for I/O. While the *de facto* standard is UTF-8, just *assuming* this can result in bugs when the code is used cross-platform. [PEP 597](https://peps.python.org/pep-0597/) addresses this, and from Python 3.10 onwards, an `EncodingWarning` *can* be raised, whenever the encoding is not specified in an `open()` statement. This should however not be an issue for FITS files, because AFAIK only ASCII is accepted for FITS anyway.

## Why this is an issue

* See [PEP 597](https://peps.python.org/pep-0597/) for details.

## Suggested implementation changes

Add the `encoding="utf-8"` keyword argument to all `open()` calls, and higher-level functions calling `open()`, such as `pathlib.Path.write_text()`.

## Resulting API changes

None, if we assume that this issues has not actually caused any practical problems for anyone *so far*.

## References

[The official PEP for this](https://peps.python.org/pep-0597/)

hugobuddel, 2023-06-13:
Yes we should absolutely do this. Otherwise files created on Windows and Linux will not be compatible with each other. But we should only do it for text files, not for FITS or binary formats.

PR #192 is intended to run the test-suite with Warnings as Errors, unless they are explicitly whitelisted. So we can have that EncodingWarning trigger an error as well. But that PR is stalled a bit and should perhaps be moved to the DevOps project.

Now all we have to do is choose an encoding, I suggest `rot-13`! J/k, rot-13 was a valid encoding in Python 2. Let's just use `utf-8`.

teutoburg, 2025-02-19:
I think most of those are done by now, so I'm closing this. If anyone finds another one, feel free to just fix it...

hugobuddel, 2025-02-21:
```
$ rg --pcre2 '^(?=.*\bopen\()(?!.*\bfits\.open\()'

docs/source/_ext/scopesim_sphinx_ext.py
27:        with open(os.path.join(output_dir, f"{eff_type}.rst"), "w") as f:
43:    with open(os.path.join(output_dir, "index.rst"), "w") as f:

docs/source/_ext/readme.txt                                                                  
12:        with open("test.txt", "w") as f:                                                  
                                                                                             
scopesim/reports/rst_utils.py
337:    with open(file_path, "w") as f:
354:    with open(file_path, "w") as f:                                                      

scopesim/server/download_utils.py                                                            
126:            with (save_path.open("wb") as file_outer,                          
                                                                                             
scopesim/server/database.py
437:    with defyam.open() as file:

scopesim/commands/user_commands.py
485:        line = open(path).readline()

scopesim/tests/tests_commands/test_UserCommands.py
131:    with open(file1, 'w') as f1:
136:    with open(file2, 'w') as f2:
141:    with open(file3, 'w') as f3:
146:    with open(file4, 'w') as f4:
152:    with open(fakelink1, 'w') as f:

scopesim/tests/tests_server/test_database.py
136:            with open(version_path) as f:
```

But I couldn't get a proper regex that also does not filter out the `encoding` line. But fine to close this