AstarVienna/ScopeSim #230: FITS headers contain no info on slit used

oczoske, 2023-06-16:
For #227  I need information on the slit that was used for the simulation (the spatial extent `xi_min` and `xi_max` or at least the slit length). In a MICADO simulation I cannot find any hint that this was a spectroscopic simulation (`DPR.TECH = IMAGE`, `WISE.INS.MODE = IMAGING`), let alone on the slit setting. 
An update is definitely needed, if only for input for pipeline development.

hugobuddel, 2023-06-16:
It is currently necessary to specify the `DPR` keywords explicitly in the OBS dictionary in order to get them in the headers, e.g. see https://github.com/AstarVienna/irdb/blob/dev_master/MICADO/docs/example_notebooks/1_scopesim_MCAO_4mas_galaxy-TLDR.ipynb

That was a very simple solution, implemented to at least have the possibility to create correct FITS files, but it is far from perfect. (It is in line with simulating dither patterns by shifting the source in the opposite direction.)

The essential problem is that the Instrument Control Software will set those header keywords, and 'simulating' the ICS is currently beyond the featureset of ScopeSim. What, in my opinion, should happen is that ScopeSim can take a full Observing Block, and then simulate all the templates therein. This would then take care of the DPR keywords, but also of dither patterns and such.

Any other headers you need can probably be added through an `ExtraFitsKeywords` keywords Effect, e.g. https://github.com/AstarVienna/irdb/blob/dev_master/MICADO/FITS_extra_keywords.yaml , which can be specified per mode, e.g. https://github.com/AstarVienna/irdb/blob/dev_master/MICADO/FITS_extra_keywords_Standalone_RO.yaml

hugobuddel, 2023-06-16:
Note that ScopeSim should have support for adding headers, but which headers to add should be specified in the IRDB, because ScopeSim is instrument-agnostic (and should even work for non-ESO telescopes).

The functionality to add header keywords is in ScopeSim. It is still preliminary, but I think it is should be sufficient for your goals. So this issue might be more relevant for the IRDB than for ScopeSim itself.

oczoske, 2023-06-16:
Yeah, it occurred to be afterwards that the irdb might have been the better place. I'll look into the `ExtraFitsKeywords` thing.

closed 2023-06-16 by oczoske