AstarVienna/ScopeSim #241: Improve base_classes

hugobuddel, 2023-07-04:
The `base_classes` (`SourceBase`, `FOVSetupBase`) seem to only exist to prevent circular imports (https://github.com/AstarVienna/ScopeSim/commit/6145ffa66707ac1d8d888b088caca44d8f6f7ea3), but they prevent any introspection by the IDE. 

For example, PyCharm complains about every `apply_to()` funtion:
```python
    def apply_to(self, obj, **kwargs):
        if isinstance(obj, SourceBase):
            for isp, spec in enumerate(obj.spectra):
                ...
```
because `SourceBase` does not have an `spectra` attribute.

It also is a code smell: apparently our code base is messy enough that we need these base classes to prevent circular imports.

What we should / could do:
* Properly document why these `base_classes` exist. The name makes it sounds like they are intended to be subclassed (beyond the classes that currently exist), but I don't think they should be.
* We should not use the base classes and import `Source` and `FovVolumeList` whenever feasible, e.g. within the `apply_to()` methods. Maybe we can then even remove `base_classes.py`.
* A bandaid: We might not want to alter the function definition of `apply_to()` (etc) everywhere. Therefor we could as a first step add an `assert isinstance(obj, Source)` where the IDE complains.
* We could perhaps add placeholders for e.g. `spectra` to the base classes. I don't think this is wise, because that's not the purpose of these classes.

@teutoburg thoughts?

teutoburg, 2023-07-04:
Hmm, this might be a hint of a larger problem lurking in the design...
I'm not a fan of the `apply_to()` function in general. It seems to do two very different things, depending on what `obj` is. Besides that, the `kwargs` don't seem to be used anywhere in the function (nor passed along to anything). But I guess they're there to not alter the parent's call signature, which is fine.
Regarding the original question: I think the first step towards resolving these base classes could be to just import the "actual" class (and perform the type check directly against that) in all cases that would not lead to circular imports anyway. I see e.g. `Source` was imported in `ter_curves.py` anyway, even before your `asserts` now...

teutoburg, 2023-07-04:
The existence of both `FieldOfViewBase` and `FOVSetupBase` seems confusing to me. Especially as both are listed as valid types for passing to the general `apply_to()` method of the `Effect` base class (which seems to be more of an actual true base class, in contrast to those others). There might have been a good reason to separate the two, but it's not immediately obvious to me.

So the `FOVManager` uses both `FieldOfView` and `FovVolumeList` objects (mind the different spelling of FOV/Fov, if nothing else), which are both valid for `apply_to()` in general, but some `Effect` subclasses (like `TERCurve`) only allow one of them (a counter-example I found is `PSF`, which allows both and deals with them differently).

It's not like I have an immediate suggestion how all of this could be done in a better way, but maybe we can come up with something in the further future 🙃

teutoburg, 2025-04-01:
I tried replacing `FOVSetupBase` with `FovVolumeList` in `TERCurve.apply_to()` (the assert with `FovVolumeList` didn't cause any issues in more than a year after all!), and now I get a circular import vom `effects.__init__.py`. Looking at the error, it seems this might be solved by not doing `from xy import *` in there, but rather explicitly import all the classes and what not. Or maybe not, idk.

More precisely, it seems that `optics.optical_element.py` imports `effects`, which in turn imports from `optics` (the `FovVolumeList`), and that from `effects` again. So perhaps rather the solution might be to split the `FOV`-related stuff from `optics` and create a new `scopesim.fovs` (better name welcome)? But I don't think this would solve this problem:

`FOVManager` needs `DetectorList` (and other effects stuff), but `TERCurve` (also an effect) needs `FovVolumeList` from `fov_manager`.

Or is it as simple as moving `FovVolumeList` to a separate module? I never liked the fact that it's stuffed in there with `FOVManager` anyway...

teutoburg, 2025-04-01:
Alright, moving `FovVolumeList` to a separate module actually solved the issue in `TERCurve `, so far so good...

teutoburg, 2025-04-01:
Turns out, `SourceBase `and `FOVSetupBase` can be removed without too much trouble. I'll see about the other ones later...

teutoburg, 2025-04-05:
Closed by #621

closed 2025-04-05 by teutoburg