AstarVienna/ScopeSim #247: PSF native resolution and Oversampling

carmeloarci, 2023-07-07:
The resolution of the PSF array used in SCOPESIM needs to properly match the Nyquist sampling of the actual PSF.
Moreover, previous experiences using SimCADO suggest avoiding intra-pixel interpolation to get rid of possible artifacts produced by numerical errors in the interpolation function. We found that running the detector images simulations with the proper combinations of pixel scale for the final detector images and for the PSF(s).
In general, we found that it is ok if the pixel scale of the final image is an integer multiple of the native resolution of the pixel scale of the PSF images  Take the MICADO case example:
MICADO has two possible pixel scales 4mas/px (zoom mode = 'wide') and 1.5mas/px ('small'). In this case, a PSF with a native resolution of 1.5 mas/px is ok for the small mode but only marginally good for the 4 mas. Otherwise, a PSF pixel scale of 0.5mas was definitely ok for both cases. 
Avoid interpolating on non-integer number of pixels (i.e. 3 to 2, instead of 4 to 2 produced better results. More specifically, since we were interested in sub-pixel accuracy for astrometric performance estimate, we ended up with PSF with pixel-scale 0.75 mas/px and optimal oversampling for the final image computation of x4 and x8 respectively for the zoom mode (0.25 max/px resolution) and for the wide (0.5 mas/px resolution.

hugobuddel, 2023-07-07:
Summarizing what we need to do:
- [ ] Add essentially @carmeloarci 's description to the documentation
- [ ] Ensure that all PSFs that are included in the IRDB follow this scheme; e.g. 0.5 mas/pixel for MICADO, unless they are only for a specific mode
- [ ] Ensure that on-the-fly generated PSFs are sampled correctly (supposedly they are already?)

oczoske, 2023-07-07:
Could you elaborate a bit on what "ok" and "better results" means and maybe provide an example to illustrate the problems? Sometimes PSFs need to be rescaled to different wavelengths (e.g. in spectroscopy) and it won't be possible to keep integer pixel scale ratios in all cases.

carmeloarci, 2023-07-07:
Hi. We were discussing some results I obtained using simcado that shares with Scopesim the use of from scipy.ndimage.interpolation import zoom in nghxrg.py . The different combination of the resolution of the PSF, of the pixelscale of the detector, and of the native resolution of the compute convolved image (and using some oversampling factor) brought to a somewhat different result in the image generated by Simcado. i.e. using wide zoom mode (4 mas/px) and a PSF of 0.75max/px and different oversampling going from 1 to 8 I found:
- native resolution 4mas px, (not ok): high flux star images presenting negative numbers
- native resolution 2mas px, (~ok): no negative numbers but some artifacts close to the PSF peak
- native resolution 1mas px, (~ok): no negative numbers much less artifacts close to the PSF peak
- native resolution 0.5mas px, (~ok): no negative numbers, no artifacts close to the PSF peak
- native resolution 0.25mas px, (~ok): no negative numbers, star image equal the one obtained at 0.5 mas.
Using the zoom mode at 1.5 mas/px and PSF at 0.75 no problem and image of the PSF constant at different oversampling modes.
Then in my jargon, "ok" means that star images at high flux are not dependent on the oversampling applied. And "better results" points out that there are numerical differences dependent on the pixel scale values introduced by the interpolation functions, but negligible from a scientific point of view.

carmeloarci, 2023-07-07:
In general, we specified that reliable PSF needs an adequate pixel scale to not produce artifacts. i.e. 2x or 3x better than Nyquist resolution.

oczoske, 2023-07-07:
Thanks. I'm not very fond of the `zoom` function, although that's more of a sentiment rather than a well-reasoned assessment. I may play around with that if I find the time.

oczoske, 2023-07-09:
Negative values can occur when the spline order is larger than 1 (cubic splines can oscillate to negative values even if the control points are all positive). The default should be linear interpolation, but can you check `cmds['!SIM.computing.spline_order']`, please?

carmeloarci, 2023-07-12:
We did an extensive test in Simcado. For the sake of the configuration of
the pixelscale of the PSF for the MORFEO/MICADO for SCOPESIM/irdb we are
recalling the point of the oversampling. in Simcado does not exist the
equivalent cmds['!SIM.computing.spline_order'](or at least I didn't find
it). However, I spotted the line of code that makes the interpolation of
the psf (in psf.py). In particular when the array of the input PSF are
even, Simcado was using spline interpolation with order 5.
This may be the cause of the negative number behaviour found so far. See
below:

def resample(self, new_pix_res):
        """
        Resample the PSF array onto a new grid

        Not perfect, but conserves flux

        Parameters
        ----------
        new_pix_res : float
            [arcsec] the pixel resolution of the returned array

        Returns
        -------
        psf_new : PSF

        Examples
        --------

            >>> new_PSF = old_PSF.resample(new_pix_res)

        """
        scale_factor = self.pix_res / np.float(new_pix_res)
        new_arr = spi.zoom(self.array, scale_factor, order=1)
        new_arr *= np.sum(self.array) / np.sum(new_arr)

        # catch any even-numbered arrays
        if new_arr.shape[0] % 2 == 0:
            tmp_arr = np.zeros((new_arr.shape[0]+1, new_arr.shape[1]+1))
            tmp_arr[:new_arr.shape[0], :new_arr.shape[1]] = new_arr
            new_arr = spi.shift(tmp_arr, 0.5, order=5)

-- 
+===========================================================
======================================
Carmelo Arcidiacono, Ph. D.

INAF Osservatorio Astronomico di Padova (OaPd)
vicolo dell'Osservatorio 5, 35122, Padova, Italy

Tel. +39 049 829 3 414 (WAS 447)
***@***.***
ORCID 0000-0003-0142-8108 <https://orcid.org/0000-0003-0142-8108>
skype: carmelo.arcidiacono
pec  : ***@***.***

Please consider the environment before printing this email.
-===========================================================
======================================

joao-aveiro, 2023-10-04:
I have just submitted Issue #282, where I conclude that bi-linear (spline_order=1) interpolation introduces a significant error in the morphology of the PSF, although in my case I am not using oversampled PSFs. 

Additionally, I have verified that in some cases the `zoom` function is not adequate for PSF resampling (see [this comment](https://github.com/AstarVienna/ScopeSim/issues/282#issuecomment-1747200849). I'll soon submit an issue on this matter with a more detailed description and some tests.

This new issue is probably not extremely pertinent to this discussion, but I thought it would be useful to link all the Issues related to PSF resampling/rescaling.