AstarVienna/ScopeSim #250: Add support for MOSAIC

hugobuddel, 2023-07-12:
We can add support for MOSAIC.

There is some work done for MOSAIC in the `kl/mos_branch` branch, https://github.com/AstarVienna/ScopeSim/tree/kl/mos_branch

Apparently this branch does not work properly at the moment.

clementhottier, 2023-07-21:
The current HEAD of `kl/mos_branch` is not functional because we tried with Kieran to imlpement a new features (Kieran told that he successfully debug the features but he didn't push it). Write now to work with Mosaic, it is mandatory to roll back to 2f0be26 for irdb (wich not used the new features)

teutoburg, 2025-08-31:
Progress for MOSAIC will be tracked in [the new corresponding Project overview](https://github.com/orgs/AstarVienna/projects/20), so I'm closing this as an individual issue.

closed 2025-08-31 by teutoburg