AstarVienna/ScopeSim #254: "Notebooks with IRDB download" test often fails with 403 error from github

hugobuddel, 2023-07-19:
The "Notebooks with IRDB download" test often, but unreliably, fails. Usually with

https://github.com/AstarVienna/ScopeSim/actions/runs/5503724179/jobs/10029211516
```
Gathering information from server ...
Connection successful, starting download ...
ERROR - HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/AstarVienna/irdb/contents/test_package?ref=dev_master (Caused by ResponseError('too many 403 error responses'))
urllib3.exceptions.ResponseError: too many 403 error responses

The above exception was the direct cause of the following exception:

[..snip..]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./docs/source/5_liners/loading_packages.py", line 63, in <module>
    sim.download_packages("test_package", release="github:dev_master")
  File "/Users/runner/hostedtoolcache/Python/3.8.17/x64/lib/python3.8/site-packages/scopesim/server/database.py", line 456, in download_packages
    pkg_path = _download_single_package(pkg_name, release, all_versions,
  File "/Users/runner/hostedtoolcache/Python/3.8.17/x64/lib/python3.8/site-packages/scopesim/server/database.py", line 346, in _download_single_package
    download_github_folder(repo_url=pkg_url, output_dir=save_dir)
  File "/Users/runner/hostedtoolcache/Python/3.8.17/x64/lib/python3.8/site-packages/scopesim/server/github_utils.py", line 82, in download_github_folder
    raise ServerError("Cannot connect to server. "
scopesim.server.github_utils.ServerError: Cannot connect to server. Attempted URL was: https://api.github.com/repos/AstarVienna/irdb/contents/test_package?ref=dev_master.
Error: Process completed with exit code 1.
```

or occasionally with

https://github.com/AstarVienna/ScopeSim/actions/runs/5515452422/jobs/10055792597
```
Testing ./docs/source/5_liners/loading_packages.ipynb
[jupytext] Reading ./docs/source/5_liners/loading_packages.ipynb in format ipynb
[jupytext] Writing ./docs/source/5_liners/loading_packages.py
Traceback (most recent call last):
  File "./docs/source/5_liners/loading_packages.py", line 43, in <module>
    sim.list_packages()
  File "/opt/hostedtoolcache/Python/3.8.17/x64/lib/python3.8/site-packages/scopesim/server/database.py", line 303, in list_packages
    all_stable = list(dict(get_all_stable(all_grouped)).keys())
  File "/opt/hostedtoolcache/Python/3.8.17/x64/lib/python3.8/site-packages/scopesim/server/database.py", line 156, in get_all_stable
    yield (package_name, get_stable(versions))
  File "/opt/hostedtoolcache/Python/3.8.17/x64/lib/python3.8/site-packages/scopesim/server/database.py", line 132, in get_stable
    return max(version for version in versions if _is_stable(version))
ValueError: max() arg is an empty sequence
Error: Process completed with exit code 1.
```

This should be resolved in some way.

teutoburg, 2024-08-03:
Haven't seen this particular one in a while. Should we close this? @hugobuddel 
Can always reopen if it returns...

hugobuddel, 2024-08-04:
Let's close this. We did make some related changes so it doesn't download things too often. Or maybe github changed some things.

closed 2024-08-04 by hugobuddel