AstarVienna/ScopeSim #280: Define milestones and projects

teutoburg, 2023-09-29:
`<disclaimer>`As I didn't find a way to create an organisation-level issue (though the request exists apparently), I'm putting this here, even though it technically concerns not only this repo.`</disclaimer>`

Today I managed to migrate the _ScopeSim-development_ project from an old ("classic") GitHub project to the new format. I choose that one, as it seemed to be the most complete one (even though _ScopeSim Complete_ exists), was still open, and also has a large number of completed things, which may be useful to show our progress (to whom it may concern...).

I would like to get rid of most if not all other old projects with "ScopeSim" in the name, as none of them seem to have been actively maintained recently anyway, and it just adds confusion. Any objections are of course welcome! After this "clean slate" status is achieved, and possibly also some closed issues and PRs from the last ~5 months added, we can re-evaluate our structure with these projects and maybe add new one(s). In any case, we should properly define what goes where, in terms of issues, PRs, projects, and finally "columns" in each project! Thoughts and suggestions very welcome!

Related, I'd like to redefine (or define at all?) our labels for issues. I added a `refactor` label and removed the repo-name-labels (`scopesim`, `scopesim-templates`, `irdb`) as they are now redundant in the new project format, which has a "repository" field. Also, we should probably be a bit more consistent / realistic with labels such as `urgent`, as there are some "urgent" issues that have been sitting around for years...

Finally, I'd like to make proper use of the "milestones" feature. We recently discussed a few deadlines and timeline-related stuff in general. @astronomyk , you seem to have the best overview of that, may I ask you to just dump them either here or in Slack, I can then put them in proper milestones here. This (c|w|sh)ould then enable us to properly use the "roadmap" feature of the project.

hugobuddel, 2023-10-02:
Link to the ScopeSim-development project: https://github.com/orgs/AstarVienna/projects/8

Feel free to remove the "ScopeSim Complete" and "Just ScopeSim" projects. I created them to experiment with the kanban boards. I did not realize it was possible to migrate the old projects, otherwise I would have done that first. I recall I couldn't even find the old projects.

I think one of the main goals of kanban boards and labels is communication, both between ourselves as developers, and with other stakeholders like the science teams. Now we are more of a team, we can make better use of github's project feature. And since you are the (at least de-facto) leader of the team, I'm happy to adapt the labels, milestones, etc. to whatever you prefer.

Rant about whether to close issues that realistically will never be fixed below.

One thing I realized is that I also track issues that I would like to get done, but realistically never will (which is fine). That is, I like to distinguish between things that we explicitly don't want to do because they are out of scope and things we implicitly don't want to do because we know they will never be important enough to prioritize them. Having that unimportant list available helps me clear my mind: as long as the never-important issue is on a list, then I don't have to think about it anymore. Or: how I learned to stop worrying and love ever-growing lists.

But for me that clarity of mind does require those unimportant issues to be 'open', or at least discoverable, that's the point. However, such issues should not clobber the backlog or lead to reoccurring discussions; the idea is to have them on a list exactly so it is not necessary to spend time on them. So I'm a bit conflicted whether to close such issues, or keep them around. Maybe a 'never' milestone, I don't know.

teutoburg, 2023-11-17:
I'll summarize here what ScopeSim-relevant "milestones" (deadlines?) I could gather from the METIS PIP-AIV google doc:

ESO Simulated data 1: 5/24 (I see no "ESO Simulated data 2", so why the "1"??)
Various input data items: 1/24, 3/24, 4/24, 5/24, 6/24 etc. from "Input data Gantt"

Assigning a milestone for each of those individual months seems overkill/cluttering. If "METIS simulated data", "ScopeSim for METIS", or similar, would be a Project (organisation-level project, to also be able to include issues and PRs in e.g. IRDB), then the "due date" could simply be a field, and only the "bigger items" would get a full milestone, i.e. "ESO Simulated data 1" and then one or two more for the remaining required input data items ("ESO Simulated data N"?). Not saying every item in that doc should become an issue, but there will certainly be some issues popping up related to those...

Thought? also @astronomyk