AstarVienna/ScopeSim #286: Check whether `sim.bug_report()` ought to be updated

oczoske, 2023-10-24:
The function `sim.bug_report()` provides version information on scopesim, python and a number of support packages. The intention was for everyone who submits a bug report to include the output of this function in the bug report. On occasion we could then simply point to this or that package and tell people to upgrade. (Getting people to actually do this is another matter...)
However, the function hasn't been updated for at least two years, as far as I can tell. If it is considered useful, could anyone have a look whether any of the packages that are currently tested need to be replaced or whether any have to be added?

teutoburg, 2023-10-24:
Yeah, I wanted to take a look at that thing for a while, but there was always something burning hotter...

hugobuddel, 2023-10-24:
Sometimes I feel like we should just include the output of `bug_report()` in every exception that is raised...

Or perhaps we could use something like Sentry (perhaps opt-in?), and catch and log every exception.

And I was wondering if there is some tool that you can give a stacktrace, and it will figure out this information based on the line numbers and such in the stack trace..

As for the function itself, it should also have information about:
- ScopeSim_Templates, if it is installed
- ScopeSim_Data, if it is installed
- Anisocado
- the IRDB, somehow (is it possible to get IRDB version info easily?)

teutoburg, 2023-10-24:
Some of this can probably be included in a reworking of logging. I started to include `logging.exception()` in some places (not only Scopesim itself), which includes the exception traceback when called inside an `except` block.

In a previous project, I included a top-level try-except around pretty much the rest of the whole code, which would log any unexpected errors. Maybe it's useful to do something similar in our high-level classes like `OpticalTrain`.

closed 2023-10-30 by teutoburg