AstarVienna/ScopeSim #295: The noise generation is not random. The seed used seems constant

carmeloarci, 2023-11-06:
I tried in different conditions and verified that if I generate two consecutive images using the same parameter I obtain exactly the same image. The background noise and the illumination Poisson noise do not change their random seed.
Attached is a notebook.
[Test random seed.zip](https://github.com/AstarVienna/ScopeSim/files/13270282/Test.random.seed.zip)

teutoburg, 2023-11-10:
This is very much by design, to have reproducable results for testing purposes!

If you include `"!SIM.random.seed": None` in your configuration, you will see a difference between each image.

closed 2023-11-10 by teutoburg

hugobuddel, 2023-11-10:
The current behavior is kinda stupid though, because all detectors get the exact same noise, see #97. Perhaps we should create an issue about how we want the RNG to behave.