AstarVienna/ScopeSim #296: In the user command the !ATMO.background and !TEL.temperature and !RO.temperature values are not considered

carmeloarci, 2023-11-06:
I tried to change independently the variable affecting the 

1. magnitude of the sky background
2. the temperature of the telescope
3. the temperature of the Relay optics

none of these affect the result.
Only the !ATMO.temperature affects the results. Please note that 2 and 3 may depend on the definition of the system in the IRDB, however, the sky background ( !ATMO.background.value) should be independent. 

Here an example, testing also random noise generation
[Test random seed.zip](https://github.com/AstarVienna/ScopeSim/files/13270469/Test.random.seed.zip)

teutoburg, 2023-11-10:
From what I could gather so far:

Elements like TEL and RO use the `!ATMO.temperature` as a default, which especially makes sense for the telescope I think. The issue is, for effects like `SurfaceList`, the values in the `.table` don't get updated after initial creation. Using your notebook's terminology, `micado2["morfeo_surface_list"].meta["temperature"]` has the final value of `9` (after running the loop 10 times), while `micado2["morfeo_surface_list"].table["temperature"]` is still all on `!ATMO.temperature`. Even setting the column in the table to the test temperature in each iteration of the loop has no effect, because e.g. `micado2["morfeo_surface_list"].surfaces["Dichroic"].meta["temperature"]` is still `!ATMO.temperature`.

This should be addressed at some point...

teutoburg, 2023-11-10:
Did some further digging, and it seems it might be as simple as changing the default `!ATMO.temperature` to `!RO.temperature` in the relay optics tables in the IRDB. The default for `!RO.temperature` itself can then point back to `!ATMO.temperature` (as is the default already!), meaning the default output will not change, _unless_ `!RO.temperature` is changed specifically. Probably a similar case for the telescope surface list and `!TEL.temperature`.

teutoburg, 2023-11-10:
For the case of `!ATMO.background`, this is most likely caused by the fact that MICADO uses a `SkycalcTERCurve` per default, and not an `AtmosphericTERCurve`, which would be affected by changing e.g. `!ATMO.background`.