AstarVienna/ScopeSim #298: Improve Random Number Generation in ScopeSim

hugobuddel, 2023-11-10:
The Random Number Generation (RNG) is a bit messy in ScopeSim. See for example #295 and #97.

The idea behind this issue is to collect what we need from the RNG.

I'll go first.

* The default should perhaps be to seed the RNG with something supposedly unique, like the MJDOBS of the simulated observation. This can be done once we can simulate full templates / observing blocks.
* If that's not possible, the default should be a 'random random number seed' I believe, unlike the current fixed seed.
* Whatever seed is used, it should be stored in the headers of the FITS file, so it is possible to uniquely recreate the exact same simulation (which also requires other keywords, e.g. for versioning).
* The RNG should be setup such that each effect 'splits of' its own generator from the main one, instead of reusing the seed. This should be easy to do in Python/numpy/scipy.
* Perhaps it should be possible to seed only specific effects, and not others. For example the generation of hot/cold pixels should perhaps be the same over many simulations, but the noise should not.

teutoburg, 2023-11-10:
I'll go next:

I recently noticed that numpy recommends using `[np.random.Generator](https://numpy.org/doc/stable/reference/random/generator.html)` for new development. AFAIK, we're using the legacy random functions most everywhere. While I didn't consider it a priority at the time I found out about this, I think if (when) we're already touching the random stuff, it might be worth migrating to the new style, which numpy claim has performance benefits, as well as quality of the "randomness" and handling. Maybe some of the changes we want to implement anyway will become easier to manage with this.

teutoburg, 2026-09-06:
I'll put this here, it could have gone into #975 as well. Tagging @hugobuddel and @oczoske to make sure you see it.

I think what we want from the random seeds is something like this:
- Default seed randomized from system entropy (aka `seed=None`), but overridable with a fixed seed for testing and pipeline data.
- Reproducible numbers for multiple readouts, ideally without depending on the previous readouts (see https://github.com/AstarVienna/ScopeSim/issues/975#issuecomment-5328317832). This means instantiating a `rng` in each effect's `__init__` won't work, because then we can only reproduce readout `n` by running readouts `1 ... n-1` before.
- Independent (but reproducible) numbers if the same effect is applied to multiple detectors.
- Independent (but reproducible) numbers between different effects that use random generators. This is less critical, because every effect does something slightly different with the `rng`, but still different seeds for different effects would be better.

I then realized these requirements are somewhat similar to the "worker ID" concept described [in the numpy docs](https://numpy.org/devdocs/reference/random/parallel.html#sequence-of-integer-seeds). Using this concept, I think we could do something like:

```python
seed = [detector_id, readout_id, effect_id, root_seed]
```

Then this would ensure independent-but-reproducible randomness between each detector, each readout, and each effect. The `root_seed` can stay the normal `"!SIM.random.seed"`, with a default to `None`. Reading the docs, this `None` should go through a `SeedSequence`, but that's an implementation detail.

The resulting seed could then be written back into the header, although I guess the only thing we really need to put in there is the resolved `root_seed`, because all the other things are deterministic anyway.

teutoburg, 2026-09-06:
Two things missing from my previous comment: The `effect_id` and `readout_id` of course don't exist yet in ScopeSim. Without going into the side rant about making ScopeSim SQL-based (I have some funny ideas), there are easy solution I think: The `effect_id` could simply be the hashed effect class name (or the `display_name` of the instance). This is easy enough to reproduce. The `readout_id` is even simpler: I'd just put a number in the `kwargs` that get propagated to the readout effects. That way, multiple readouts can `enumerate` the loop, and a single one can either default to `0` or use a specific number to reproduce a specific readout, without the need to run the previous ones.

hugobuddel, 2026-09-06:
Sounds good!

Pro's:

- All the seeds can be calculated in advance (right?), so it is easily parallelizable.  (But some effects modify the optical train right? Is that a problem?)
- When an effect is added or removed but the root seed is kept, then the seed for each effect is unaffected.  (So one can truly see the effect each effect has.)
- All information can be put in the FITS headers easily: the root seed and the `readout_id` on the top level.  And we could store the `detector_id` in each extension, and HIERARCH keywords to store the `effect_id` and final `seed` for each individual effect.
- Then we can also turn those header keywords back into seeds to provide ScopeSim.  The `root_seed` and `readout_id` would be easy, and maybe all that is needed initially.

Some thoughts:

- You wouldn't always want to vary by `readout_id`.  For example, if the bad pixels are determined randomly, then you'd want those to be the same between readouts, but not per detector.

And we should put the full effect configuration in a database; I already did that for MICADO.  The full circle: ScopeSim input yaml -> FITS headers -> Database -> ScopeSim input yaml.  (We should also put the effect code in a database, so people can code up their own effects and let other users query for them. )  (But we should use a graph database, not an SQL one.)

teutoburg, 2026-09-06:
> And we should put the full effect configuration in a database; I already did that for MICADO. The full circle: ScopeSim input yaml -> FITS headers -> Database -> ScopeSim input yaml. (We should also [...]

Gives us something to do for the next 4 years 😆 

I have little experience with graph databases, hence my mention of SQL. I had that vague idea for a while now, it always felt interesting but a little bit off, so the graph thing is probably the right answer. My basic idea was that the whole OpticalTrain config could be mapped as relations. This would also solve the issue of effects needing to access their parent OpticalTrain, or other effects in the same, which is currently very cumbersome, but could become just a matter of following references. Which, to be fair, does sound already a lot like traversing nodes in a graph...

teutoburg, 2026-09-06:
> * You wouldn't always want to vary by `readout_id`.  For example, if the bad pixels are determined randomly, then you'd want those to be the same between readouts, but not per detector.

Sure, in my draft, the basic thing would be `[effect_id, root_seed]`, to which the individual effects can prepend (numpy docs recommend putting the root seed last) the `detector_id` and/or `readout_id` as needed, or don't.

> * All the seeds can be calculated in advance (right?), so it is easily parallelizable.  (But some effects modify the optical train right? Is that a problem?)

Yes, this should be deterministic given the `root_seed`. I don't worry so much about parallelization in the `readout` loop, because that's fairly cheap compared to the `observe`, so unless we read out more than ~ 20-50 times on the same observation, might not be that beneficial to parallelize. Something like the spectral traces are a more promising candidate. But in any case, this proposed seed scheme shouldn't get in the way of any potential parallelization, after all the relevant numpy docs make most of their exapmles about that...