AstarVienna/ScopeSim #366: Make sure all core functionality is implemented (e.g. MOS)

teutoburg, 2024-02-07:


hugobuddel, 2024-02-08:
Adding MOS functionality seems like a huge thing to do before 1.0!

Would adding MOS functionality require fundamental changes to the inside of ScopeSim in some way? As in, would adding the MOS functionality later require a big rewrite that would break backwards compatibility?

teutoburg, 2024-02-08:
> Adding MOS functionality seems like a huge thing to do before 1.0!
> 
> Would adding MOS functionality require fundamental changes to the inside of ScopeSim in some way? As in, would adding the MOS functionality later require a big rewrite that would break backwards compatibility?

@astronomyk seems to be pushing for the MOS stuff to be merged in, so I added it here. I'm unsure about priorities and consequences at this point (see the aftermath of the currsys)...