AstarVienna/ScopeSim #388: Create top-level user interface

teutoburg, 2024-03-07:


teutoburg, 2024-03-07:
Thoughts, @astronomyk and @hugobuddel ?

I suppose this is a useful thing to get away from some of the recent "developer's happy time" (😇) and add something that will actually be useful to the end user. And tick a box on the long and [arduous march](https://en.wikipedia.org/wiki/North_Korean_famine#Arduous_March) towards v1.0, that I had as a draft issues apparently since October 😅

closed 2024-09-30 by teutoburg

hugobuddel, 2024-10-02:
> Come to think of it, I (or someone) should really write some [user stories](https://en.wikipedia.org/wiki/User_story) on all of this 🙃

I came back to this issue, because maybe it should only be closed once there is proper documentation?

Maybe all PRs should have documentation associated with it.