AstarVienna/ScopeSim #408: Problems with MICADO simulations adopting a real image as input

fannibali, 2024-04-29:
Hi, I am struggling with simulating MICADO images where my input is a real HST image. 
I do the following: 

**Adopt my HST image** 
fits_image_filename = "j9fv02xqq_drz.fits"
hdu = fits.open(fits_image_filename)

**Delete the image header and replace with new WCS info, for instance:**

del hdu[1].header[0:-1]

hdu[1].header.update({"CDELT1": 0.018, "CUNIT1": "arcsec", "CRPIX1": 581, "CRVAL1": 0,
                   "CDELT2": 0.018, "CUNIT2": "arcsec", "CRPIX2": 568, "CRVAL2": 0,
                    "NAXIS1": 1163, "NAXIS2" : 1136, "NAXIS" : 2})

image_source = scopesim.Source(image_hdu=hdu[1], flux=22*u.ABmag)

....
....
 launch micado simulation, this is the result

[image]

However, if  **I decrease  the pixel scale in the header,**  for instance CDELT1": 0.015 instead of 0.018, I just get **an empty image.** This seems to happen for cdelt>0.017". High cdelt values seem on the other hand to work properly. 

So I guess there is a bug related to how scopesim interpret the image header?

hugobuddel, 2024-04-29:
Could you please share the input image that you are using @fannibali ? That makes it easier to reproduce exactly the problem you are having.

fannibali, 2024-04-30:
Hi,

I am trying to upload the fits HST image but it is too big...
Is there a drive where I can upload it?
Otherwise, I could share a google drive with the image with some e-mail
address

just let me know
Francesca

fannibali, 2024-04-30:
Hi,

here is the link to the drive where I have uploaded the image

https://drive.google.com/file/d/1i63Ng59rcE3YIv5ASUE9u7rOSy653GbB/view?usp=sharing

Let me know if you have problems in accessing it

many thanks in advance!
Francesca

hugobuddel, 2024-05-01:
Thansk @fannibali , I've reproduced the problem.

I've uploaded the file as a zipfile to the issue:  [j9fv02xqq_drz.fits.zip](https://github.com/AstarVienna/ScopeSim/files/15177951/j9fv02xqq_drz.fits.zip)

```python
import scopesim
from astropy.io import fits
from astropy import units as u
import datetime

fits_image_filename = "j9fv02xqq_drz.fits"
hdu = fits.open(fits_image_filename)

del hdu[1].header[0:-1]

hdu[1].header.update({"CDELT1": 0.018, "CUNIT1": "arcsec", "CRPIX1": 581, "CRVAL1": 0,
"CDELT2": 0.018, "CUNIT2": "arcsec", "CRPIX2": 568, "CRVAL2": 0,
"NAXIS1": 1163, "NAXIS2" : 1136, "NAXIS" : 2})

# Adding this line gives an image with only readout noise it seems.
# Commenting it out will show the source.
hdu[1].header.update({"CDELT1": 0.015, "CDELT2": 0.015})

image_source = scopesim.Source(image_hdu=hdu[1], flux=22*u.ABmag)

cmd = scopesim.UserCommands(use_instrument="MICADO", set_modes=["MCAO", "IMG_4mas"])
cmd.update(properties={
    "!OBS.filter_name_fw1": "open", 
    "!OBS.filter_name_fw2": "H", 
    "!OBS.ndit": 20,
    "!OBS.dit": 3,

    "!OBS.catg": "SCIENCE",
    "!OBS.tech": "IMAGE",
    "!OBS.type": "OBJECT",
    "!OBS.mjdobs": datetime.datetime(2022, 1, 1, 2, 30, 0)
})

cmd["!DET.width"] = 4096     # pixel
cmd["!DET.height"] = 4096

micado = scopesim.OpticalTrain(cmd)
micado.observe(image_source)
hdus = micado.readout()

%matplotlib inline
import matplotlib.pyplot as plt
from matplotlib.colors import LogNorm

plt.figure(figsize=(10,10))
plt.imshow(hdus[0][1].data, norm=LogNorm(vmin=1e3))
plt.colorbar()
```

Result with a pixel scale of 0.018:

[image]

Result with a pixel scale of 0.015:

[image]

I have not yet investigated the cause. @teutoburg do you have an opportunity to look at this?

Two things I noticed:
- The source image is only 1136 pixels in size, and it is rendered on a 4k detector, so it is severely undersampled. That should work, but maybe that is causing problems? I think we usually have oversampled sources, but not sure. And this would not explain why it works at some scales but not the other.
- The source image has NaN's in it, maybe that causes problems? But then I would expect that problem to occur earlier. Or maybe the NaNs are out of the field of view at 0.015 and inside the fov at 0.018?

hugobuddel, 2024-05-01:
@teutoburg I suppose we could have ScopeSim do something more sensible in the presence of NaNs right? E.g. either fill them in with the median value, or treat them as transparant. At the very least it would be good to give the user a warning instead of (effectively) silently discarding the entire source.

teutoburg, 2024-05-01:
> @teutoburg I suppose we could have ScopeSim do something more sensible in the presence of NaNs right? E.g. either fill them in with the median value, or treat them as transparant. At the very least [...]

Yes. At the very least check for NANs and log a Warning. Maybe look into masked arrays for dealing with them...

fannibali, 2024-05-02:
Many thanks indeed!
I will try to implement your solution and let you know if I encounter additional problems!
Thanks!!!
Francesca

closed 2024-05-23 by hugobuddel