AstarVienna/ScopeSim #419: have ScopeSim do something more sensible in the presence of NaNs

hugobuddel, 2024-05-23:
>  @teutoburg I suppose we could have ScopeSim do something more sensible in the presence of NaNs right? E.g. either fill them in with the median value, or treat them as transparant. At the very least it would be good to give the user a warning instead of (effectively) silently discarding the entire source.

Yes. At the very least check for NANs and log a Warning. Maybe look into masked arrays for dealing with them...

_Originally posted by @teutoburg in https://github.com/AstarVienna/ScopeSim/issues/408#issuecomment-2088625718_

teutoburg, 2024-05-23:
> Maybe look into masked arrays for dealing with them...

Yeah, those will probably be useful anyway...

hugobuddel, 2024-05-23:
At some point we'd need to convert the NaN to a value to include on the detector, irrespective of whether they are masked.

The user that created the source object with NaNs is in the best position to decide what a NaN mean. We cannot do that for them. So I think we should formally say that there must be a value for each pixel, and never NaN.

But the user probably just didn't realize there are NaNs, so as a courtesy we can do something sensible. I think for all practical purposes a NaN in a source pixel means 0 flux.

So I propose we do two things:
- Warn the user that there are NaN's and that they should fix those before using it as input to ScopeSim.
- Convert all NaN's in a source object to 0 anyway.

I'd also be fine with:
- Having a flag somewhere to explicitly convert NaN values to 0s (or some other value).
- Raising an error on NaN values, pointing the user to that flag.

Or just:
- Raising an error on NaN values.

The latter might be the easiest to implement :safety_vest: .

teutoburg, 2024-05-23:
> * Having a flag somewhere to explicitly convert NaN values to 0s (or some other value).

I'd go for that eventually. Adding a Warning should be easy to do. Though I won't get around to it until next week...

PR #466 (2024-09-06): Replace NaNs in images, log warning
Closes #419 

Simple example to illustrate:
```python
import numpy as np
from matplotlib import pyplot as plt
import scopesim as sim
import scopesim_templates as sim_tp

src=sim_tp.galaxy("brown/NGC5866")
src.fields[0].data[70:80, 70:80] = np.nan
np.isnan(src.fields[0].data).sum()

sim.download_packages(["ELT", "Armazones", "MICADO"])
cmds=sim.UserCommands(use_instrument="MICADO")
opt=sim.OpticalTrain(cmds)

opt.observe(src)
plt.imshow(opt.readout()[0][1].data)
```

shows:

[image]

and prints:

```
astar.scopesim.optics.fov - WARNING: Source contained NaN v [... cut]
closed 2024-09-06 by teutoburg