AstarVienna/ScopeSim #430: Observation result in a empty image

AlessioTar, 2024-06-19:
I'm having hard times trying to simulate the observation of an artificial image, generated with Galfit, using scopesim. It result in an empty image with only the noise inside. I use the same code to produce simulations of other targets and it works well so I'm a bit surprised and confused by this particular behavios, can it be something linked to the specific file I'm using or am I missing something?

I attach the file mentioned above and the code I use, as well as the result I obtain.

import numpy as np
import scipy
import astropy.io.fits as fits
from astropy import units as u
import matplotlib.pyplot as plt
from matplotlib.colors import LogNorm
%matplotlib inline

import scopesim

fits_image_filename = "../gal_fit/slice-clump-orig.fits"    #modify this for portability
hdu = fits.open(fits_image_filename)
del hdu[0].header[0:-1]

# Give the header some proper WCS info
hdu[0].header.update({"CDELT1": 0.004, "CUNIT1": "arcsec", "CRPIX1": 512, "CRVAL1" :0,
                   "CDELT2": 0.004, "CUNIT2": "arcsec", "CRPIX2": 512, "CRVAL2" : 0,
                    "NAXIS1": 1024, "NAXIS2" : 1024, "NAXIS" : 2})

hdu[0].header
mask = np.isnan(hdu[0].data)
tofill = np.median(hdu[0].data[~mask])
hdu[0].data[mask] = tofill
image_source = scopesim.Source(image_hdu=hdu[0], flux=26.4*u.ABmag)

def plot(src):
    plt.figure(figsize=(12, 5))
    plt.subplot(121)
    wave = range(7000, 23000)
    plt.plot(wave, image_source.spectra[0](wave))
    plt.xlabel("Wavelength [Angstrom]")
    plt.ylabel("Flux [ph/s/cm2/Angstrom]")
    plt.subplot(122)
    plt.imshow(np.abs(image_source.fields[0].data),origin='lower')
    plt.colorbar()

plot(image_source)

TEMP_ZERO = 5.0 # REFERENCE TEMP
TEMP_CELSIUS = 5.0 # AMBIENT TEMP
MAORY_TEMP   = 5.0 # MORFEO TEMP
DIT = 1000 # DIT in seconds
NDIT = 18 # NDIT in seconds
oversampling = 4
s_filter = "I-long" # I-long, Y, J, H, Ks
# set I band configuration
mode = "MCAO"
#-----------------------------------------------------------------------------------------------
cmd = scopesim.UserCommands(use_instrument="MICADO", set_modes=[mode, "IMG_4mas"])
cmd.update(properties={
    "!OBS.filter_name_pupil": "open",
    "!OBS.filter_name_fw1": s_filter, # I,J,H
    "!OBS.filter_name_fw2": "open",  # Ks
    "!OBS.ndit": NDIT, 
    "!OBS.dit": DIT,
    "!OBS.instrument": "MICADO",
    "!OBS.catg": "SCIENCE",
    "!OBS.tech": "IMAGE",
    "!OBS.type": "OBJECT",
    "!OBS.airmass": 1.0,
    "!ATMO.background.filter_name": s_filter,
    "!ATMO.temperature": TEMP_CELSIUS,
    "!TEL.temperature":TEMP_CELSIUS,
    "!SIM.oversampling":oversampling,
    "!SIM.verbose":True,
    "!SIM.random.seed": None,
    "!DET.width": 1024,     # pixel central detector
    "!DET.height": 1024})   # pixel central detector
micado = scopesim.OpticalTrain(cmd)
from scopesim.effects.psfs import FieldConstantPSF
micado["dark_current"].include = True
micado["shot_noise"].include = True 
micado["detector_linearity"].include = False
micado["exposure_action"].include = True
micado['readout_noise'].include = True
# selecting the PSF desired
micado["morfeo_generic_psf"].include = True
micado.observe(image_source)
hdus_I = micado.readout(filename= "./sim_fits/scopesim_clump.fits")
im_I= hdus_I[0][1].data
fig,axes = plt.subplots(nrows=1, ncols=1, figsize=(12,10), sharex=True, sharey=True)
im = axes.imshow(im_I,  cmap="inferno")
axes.set_title('K-band MORFEO "PSF 2023" ')

fig.colorbar(im, ax=axes, label='Interactive colorbar')
axes.set_xlim([0, 1024 ])
axes.set_ylim([0, 1024 ])
plt.show()

At this point I expect to see something in the image other than the noise, I also tried to increase the value of the flux associated to the image hoping for something to pop up but it didn't work.
At the following link you should be able to download the fits file I use as input since the format is not supported on github.
https://mega.nz/file/oWcBHIxB#WKzi2WifK2KRht4vpkUKaFNwCmgeGfmaeK5Tc8DQmdE

I recall that the same exact code works with other input files so (I guess) the problem is about this specific file.

hugobuddel, 2024-06-20:
I think the code kinda works as intended. The `flux` argument only scales a standard star spectrum, and uses that as the spectrum for the image. The pixel values are not considered at all when handling the `flux` argument.

So the pixel values should still be set to a meaningful value. So instead of the above changes, you can also add just this line:
```python
hdu[0].data *= 1e36
```
(which is a larger multiplication than above).

hugobuddel, 2024-06-20:
Perhaps an action item on us is to properly document how all these numbers tie together. E.g. it should be clear what value of `flux` and pixel values in the input image should lead to what pixel values in the output image (e.g. assuming `CDELT` of both the input and output images are equal).

AlessioTar, 2024-06-20:
Hi @hugobuddel unfortunately removing the `flux` argument from `scopesim.Source()` raise an error:
**image_hdu must be accompanied by either spectra or flux** so I think I have to keep it in my code.

For what regards the image you show i was expecting something like this, thank you very much. The only problem is the scale you obtain, the objects you see in my input image are very faint objects (around 28-29 mag) and I expect to see them immerse in the noise and barely detectable.

Working with your suggestions I figured out that I was mixing different units and this caused my problem, the input image is in units of erg/s/cm^2/Hz while (as far as I know) the system produces the images and the noise in units of electron counts. Do you know any smart way to convert my image in counts itself so that it can be compared with the noise?

hugobuddel, 2024-06-20:
> Hi @hugobuddel unfortunately removing the `flux` argument from `scopesim.Source()` raise an error: **image_hdu must be accompanied by either spectra or flux** so I think I have to keep it in my code.

In my first suggestion, I changed four lines, all marked with my initials `HB`. One of those lines is adding a `BUNIT` to the header, which allows you to remove the `flux` argument. I expect that you should get things working properly if you figure out what to put into `BUNIT`.
 
> For what regards the image you show i was expecting something like this, thank you very much. The only problem is the scale you obtain, the objects you see in my input image are very faint objects (around 28-29 mag) and I expect to see them immerse in the noise and barely detectable.

Yeah, I just scaled them arbitrarily until they were visible, making sure that ScopeSim is at least doing _something_. As I was afraid that there was some problem that prevented anything from showing at all.

> Working with your suggestions I figured out that I was mixing different units and this caused my problem, the input image is in units of erg/s/cm^2/Hz while (as far as I know) the system produces [...]

I'm not sure how much help I can be w.r.t. to how fluxes are dealt with in ScopeSim, as I have not spent enough time on that part of the code. However, with the information you have given, I've just done this:

```python
hdu[0].header["BUNIT"] = "erg/s/cm^2/Hz"
```
That still didn't show a picture that I could visually distinguish any source. However, increasing the flux by only 1 magnitude (a factor of 2.5) already shows something (see image below):
```python
hdu[0].data *= 2.5
```
So maybe setting the BUNIT would already make the source detectable with the proper tools.

To be clear: I don't know for sure whether setting `BUNIT` is really the correct way to do things, but it seems to work out well.

[image]

AlessioTar, 2024-06-22:
Thank you very much Hugo, with your support and your hints I managed to solve the problem.

closed 2024-06-22 by AlessioTar