AstarVienna/ScopeSim #475: Scopesim installation with pip does not work out of the box

jhartke, 2024-10-14:
Hi, 
I tried to install scopesim with `pip install scopesim` in a fresh conda environment with only pip (24.2) and python (3.13.0). The installation failed with the error message below. I could work around this by installing astropy=5.3.4 with conda and then pip install worked without issues. 
Thanks for looking into this!
Best,
Johanna

`Collecting astropy<6.0.0,>=5.3.4 (from scopesim)
  Using cached astropy-5.3.4.tar.gz (7.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [32 lines of output]
      Traceback (most recent call last):
        File "/opt/homebrew/anaconda3/envs/scopesim_test/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
          ~~~~^^
        File "/opt/homebrew/anaconda3/envs/scopesim_test/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "/opt/homebrew/anaconda3/envs/scopesim_test/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/private/var/folders/kj/cb8jn769595d_tgtdx98315w0000gp/T/pip-build-env-clkwl4ni/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/kj/cb8jn769595d_tgtdx98315w0000gp/T/pip-build-env-clkwl4ni/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 302, in _get_build_requires
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "/private/var/folders/kj/cb8jn769595d_tgtdx98315w0000gp/T/pip-build-env-clkwl4ni/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 318, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 69, in <module>
        File "/private/var/folders/kj/cb8jn769595d_tgtdx98315w0000gp/T/pip-build-env-clkwl4ni/overlay/lib/python3.13/site-packages/extension_helpers/_setup_helpers.py", line 63, in get_extensions
          for setuppkg in iter_setup_packages(srcdir, packages):
                          ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/kj/cb8jn769595d_tgtdx98315w0000gp/T/pip-build-env-clkwl4ni/overlay/lib/python3.13/site-packages/extension_helpers/_setup_helpers.py", line 159, in iter_setup_packages
          module = import_file(setup_package, name=packagename + ".setup_package")
        File "/private/var/folders/kj/cb8jn769595d_tgtdx98315w0000gp/T/pip-build-env-clkwl4ni/overlay/lib/python3.13/site-packages/extension_helpers/_utils.py", line 136, in import_file
          loader.exec_module(mod)
          ~~~~~~~~~~~~~~~~~~^^^^^
        File "<frozen importlib._bootstrap_external>", line 1022, in exec_module
        File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
        File "./astropy/wcs/setup_package.py", line 13, in <module>
          from setuptools.dep_util import newer_group
      ModuleNotFoundError: No module named 'setuptools.dep_util'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.`

oczoske, 2024-10-14:
This is certainly related to https://github.com/astropy/astropy/pull/15634
I don't think the fix has made it into astropy 5.3.4 but is only available for >= 6.0.

teutoburg, 2024-10-14:
ScopeSim v0.9 (to be released this month) will require Astropy >= 6.0.1, which will hopefully fix this. Also keep in mind that ScopeSim is currently not tested for Python 3.13 (we're working on that), so using a Python version outside the 3.10-3.12 range may have unexpected consequences.

closed 2024-12-19 by oczoske