AstarVienna/ScopeSim #488: METIS/LSS_AGN-02_simulation.ipynb is broken

teutoburg, 2024-11-03:
The result using current dev versions now looks like this:

[image]

For comparison, since the notebook does not have an output saved in the IRDB, here's a run from an older RTD build where it looks more reasonable: https://irdb.readthedocs.io/en/fh-rtd/METIS/docs/example_notebooks/LSS_AGN-02_simulation.html

Not sure what's causing this or since when it occurs. It was not a result of #482, it was already the same using `main` before that merge.

I put this here even though it's an IRDB notebook, because I belief the issue to be within ScopeSim and probably also affects other things.

hugobuddel, 2024-11-03:
- ScopeSim `v0.7.0`
- ScopeSim_Templates `v0.5.4`
- IRDB `METIS.2024-02-29`

Still looks OK:

[image]

hugobuddel, 2024-11-03:
According to git bisect:
```
423ba0c3d259a53be5c714c3f917e42b91f9735f is the first bad commit
commit 423ba0c3d259a53be5c714c3f917e42b91f9735f
Author: teutoburg <ghost@instruct.at>
Date:   Thu Apr 11 15:12:24 2024 +0200

    Only auto-determine dit, ndit when not given
    
    Plus some extras to make sure edge cases are covered.
    Use meaningful default if no exptime given.
    Also include lots of logging.

 scopesim/effects/electronic.py | 130 ++++++++++++++++++++++++++++-------------
 1 file changed, 89 insertions(+), 41 deletions(-)
```
Maybe you can take it from here?

This was with irdb version:
```
git checkout e60a3b86~
```

scopesim.bug_report():
```
Python:
3.11.5 | packaged by conda-forge | (main, Aug 27 2023, 03:34:09) [GCC 12.3.0]

Installed Python packages:
           anisocado: 0.3.0
         anisocado (: could not be loaded.
       astar-utils (: could not be loaded.
           astropy (: could not be loaded.
    beautifulsoup4 (: could not be loaded.
          docutils (: could not be loaded.
             httpx (: could not be loaded.
              lxml (: could not be loaded.
        matplotlib (: could not be loaded.
    more-itertools (: could not be loaded.
             numpy (: could not be loaded.
             pooch (: could not be loaded.
            pyyaml (: could not be loaded.
             scipy (: could not be loaded.
       scopesim_data: 0.0.1a0
  scopesim_templates: 0.5.5a0
       skycalc-ipy (: could not be loaded.
           synphot (: could not be loaded.
              tqdm (: could not be loaded.

Installed IRDB packages:
                .git: version number not available.
             .github: version number not available.
           Armazones: 2023-07-11
                 ELT: 2023-07-11
                 GTC: version number not available.
               HAWKI: 2023-06-14
                 HST: version number not available.
                LFOA: 2022-04-12
             LaPalma: version number not available.
               MCIFU: version number not available.
               METIS: 2023-07-11
              MICADO: 2023-07-14.dev
          MICADO_Sci: version number not available.
                NACO: version number not available.
              OSIRIS: version number not available.
             Paranal: 2022-04-09
                 VLT: version number not available.
            ViennaLT: version number not available.
                WFC3: version number not available.
            _REPORTS: version number not available.
                docs: version number not available.
                irdb: version number not available.
        test_package: 2023-07-20.dev
               .idea: version number not available.
       .pytest_cache: version number not available.
  .ipynb_checkpoints: version number not available.
                ERIS: version number not available.
                FORS: version number not available.
             HARMONI: version number not available.
                KMOS: version number not available.
              MORFEO: 2023-07-11
                MUSE: version number not available.
                UVES: version number not available.
               VISIR: version number not available.
    _ZIPPED_PACKAGES: version number not available.
              DREAMS: version number not available.

Operating System info:
   System: Linux
  Release: 5.4.0-198-generic
  Version: #218-Ubuntu SMP Fri Sep 27 20:18:53 UTC 2024
  Machine: x86_64
```

teutoburg, 2024-11-03:
> Maybe you can take it from here?

I belief so, yes. Thanks a lot for doing the legwork here!!

hugobuddel, 2024-11-03:
You need to add
```
                                  "!OBS.ndit": 100350,
                                  "!OBS.dit": 0.036,
```
to the UserCommands, that is what the parent of 423ba0c3d259a53be5c714c3f917e42b91f9735f calculates, then you will get a signal

But now I just get
```
ValueError: HDUList is required to retrieve -TAB coordinates and/or indices.
```

I'm off to bed.

hugobuddel, 2024-11-04:
One thing that we did not properly test in #428 is the role of MINDIT, because both MINDIT and DIT are set to 1, so we don't know whether the AutoExposure is working as intended.

Also, if I add a `print(repr(opt[3][2]))` to the right place in `test_autoexp_overrides_obs_dict()`:
```
    def test_autoexp_overrides_obs_dict(self, obs_aeq, exptime, factor, quant):
        """This should prioritize kwargs and use dit, ndit when None."""
        opt, default, quanteff = obs_aeq
        kwarged = int(opt.readout(exptime=exptime)[0][1].data.sum())
        print(repr(opt[3][2]))
```

It will print out 
```
astar.scopesim.effects.electronic - Exposure parameters: DIT = 10.000 s, NDIT = 3
astar.scopesim.effects.electronic - Total exposure time: 30.000 s
AutoExposure(**{'filename': None, 'description': '', 'history': [], 'name': '<empty>', 'mindit': 1, 'full_well': 100, 'fill_frac': 0.8, 'z_order': [902], 'include': True, 'cmds': UserCommands(**{'use_instrument': 'basic_instrument', 'ignore_effects': ['shot_noise', 'dark_current', 'readout_noise', 'atmospheric_radiometry', 'source_fits_keywords', 'effects_fits_keywords', 'config_fits_keywords'], 'properties': {'!OBS.dit': 10, '!OBS.ndit': 1}})})
```

See the properties part at the end: it seems that the calculated dit and ndit are not stored, so it seems we don't even have a way to figure out what the calculated dit/ndit are?

teutoburg, 2024-11-04:
> See the properties part at the end: it seems that the calculated dit and ndit are not stored, so it seems we don't even have a way to figure out what the calculated dit/ndit are?

I belief the properties shown when printing out the UserCommands object are those that were used initially to create the instance.