AstarVienna/ScopeSim #491: sim.bug_report() failing

AGlasse, 2024-11-05:
This is my first METIS bug report.  For information, I am running ScopeSim v0.8.4 and ScopeSim_templates v0.5.4 using the 'PyCharm' IDE.  I have a python script (starting fragment below) which runs end-to-end to generate a METIS IFU spectral images.  However, if I insert a call to sim.bug_report() at the start it crashes, 
-------------------------------------------
#!/usr/bin/env python
import numpy as np
import scopesim as sim
import scopesim_templates as sim_tp
from matplotlib import pyplot as plt
import pyckles
from astropy.io.fits import BinTableHDU
from astropy import units as u
from synphot import models, SourceSpectrum, Empirical1D

sim.bug_report()
----------------------------------------------

output
-------------------------------------------
C:\Users\hkn21281\AppData\Local\Programs\Python\Python311\python.exe D:\Users\hkn21281\PycharmProjects\metsim\source\metsim.py 
Python:
3.11.1 (tags/v3.11.1:a7a450f, Dec  6 2022, 19:58:39) [MSC v.1934 64 bit (AMD64)]

Installed Python packages:
Traceback (most recent call last):
  File "D:\Users\hkn21281\PycharmProjects\metsim\source\metsim.py", line 11, in <module>
    sim.bug_report()
  File "C:\Users\hkn21281\AppData\Local\Programs\Python\Python311\Lib\site-packages\scopesim\utils.py", line 200, in bug_report
    _write_bug_report(sys.stdout)
  File "C:\Users\hkn21281\AppData\Local\Programs\Python\Python311\Lib\site-packages\scopesim\utils.py", line 168, in _write_bug_report
    packages = set(_get_required_packages())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hkn21281\AppData\Local\Programs\Python\Python311\Lib\site-packages\scopesim\utils.py", line 139, in _get_required_packages
    for req in reqs:
TypeError: 'NoneType' object is not iterable
----------------------------------------------------

END Glasse 5/11/24

hugobuddel, 2024-11-05:
Thanks @AGlasse. Ironic that the bug reporting feature contains a bug...

(I've taken the liberty to replace your dashed lines with three backticks. Github uses markdown, and a dashed line means that the line above it is a header, while text enclosed between lines with three backticks means something like verbatim in latex.)

hugobuddel, 2024-11-05:
It should be easy to add a fail safe for this error, but I don't really understand what is going on. It looks like there is something weird with your installation.

Is there a directory "C:\Users\hkn21281\AppData\Local\Programs\Python\Python311\Lib\site-packages\scopesim-0.8.4.dist-info" on your system (or similar perhaps)? And does that have a file "METADATA" in it? If so, could you please share the contents of that file?

AGlasse, 2024-11-05:
Hi Hugo,

  I’ve attached the file to this message.  I will be off-line for the next couple of hours,

  Thank you for your help,
  ..Alistair

Alistair Glasse
JWST-MIRI Instrument Scientist, ELT-METIS CoI
UK Astronomy Technology Centre
The Royal Observatory, Edinburgh
***@***.******@***.***>

UK-(0)-131-385-9399  Zoom
UK-(0)-7887-386-997  Mobile

Metadata-Version: 2.1
Name: ScopeSim
Version: 0.8.4
Summary: Generalised telescope observation simulator
Home-page: https://github.com/AstarVienna/ScopeSim
License: GPL-3.0-or-later
Author: Kieran Leschinski
Author-email: ***@***.***
Maintainer: Kieran Leschinski
Maintainer-email: ***@***.***
Requires-Python: >=3.9,<4.0
Classifier: Intended Audience :: Science/Research
Classifier: License :: OSI Approved :: GNU General Public License v3 or later (GPLv3+)
Classifier: Operating System :: OS Independent
Classifier: Programming Language :: Python :: 3
Classifier: Programming Language :: Python :: 3.9
Classifier: Programming Language :: Python :: 3.10
Classifier: Programming Language :: Python :: 3.11
Classifier: Programming Language :: Python :: 3.12
Classifier: Topic :: Scientific/Engineering :: Astronomy
Requires-Dist: anisocado (>=0.3.0)
Requires-Dist: astar-utils (>=0.2.2)
Requires-Dist: astropy (>=5.3.4,<6.0.0)
Requires-Dist: beautifulsoup4 (>=4.12.1,<5.0.0)
Requires-Dist: docutils (>=0.19,<0.20)
Requires-Dist: httpx (>=0.23.0,<0.24.0)
Requires-Dist: lxml (>=4.9.3,<5.0.0)
Requires-Dist: matplotlib (>=3.8.2,<4.0.0)
Requires-Dist: more-itertools (>=10.1.0,<11.0.0)
Requires-Dist: numpy (>=1.26.3,<2.0.0)
Requires-Dist: pooch (>=1.8.2,<2.0.0)
Requires-Dist: pyyaml (>=6.0.1,<7.0.0)
Requires-Dist: scipy (>=1.11.4,<2.0.0)
Requires-Dist: skycalc_ipy (>=0.4.0)
Requires-Dist: synphot (>=1.3.post0,<2.0)
Requires-Dist: tqdm (>=4.66.1,<5.0.0)
Project-URL: Bug Tracker, https://github.com/AstarVienna/ScopeSim/issues
Project-URL: Documentation, https://scopesim.readthedocs.io/en/latest/
Project-URL: Repository, https://github.com/AstarVienna/ScopeSim
Description-Content-Type: text/markdown

# ScopeSim 
## A telescope observation simulator for Python

[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]

[image]
[image]

## Summary

ScopeSim aims to simulate images of astronomical objects observed with visual 
and infrared instruments. It does this by creating models of the optical train 
and astronomical objects and then pushing the object through the optical train. 
The resulting 2D image is then broadcast to a detector chip and read out into a 
FITS file. 

This code was originally based on the [SimCADO](https://github.com/astronomyk/simcado) package

## Documentation
The main set of documentation can be found here: 
https://scopesim.readthedocs.io/en/latest/

A basic Jupyter Notebook can be found here: 
[scopesim_basic_intro.ipynb](docs/source/examples/1_scopesim_intro.ipynb)

hugobuddel, 2024-11-05:
Thanks for sharing the file. The files looks all normal, but it seems that for some reason it cannot be found by `importlib.metadata`. (I tried to reformat the comment, but apparently failed.)

I've added a workaround for the bug in #492, so that at least `bug_report()` wouldn't crash, but there is still something weird going on. The fix should make it to ScopeSim v0.9.0 I expect, which we plan to release soon.

I suggest to try things with a fresh installation, especially if you face more of such problems. For example, you are still using Python 3.11.1, while 3.11.10 is the latest.

PR #492 (2024-11-05): Workaround for #491 metadata.requires can return None
Closes #491 . Or well, ensures that `bug_report()` does not crash. We could perhaps do something smarter like log an error or so, maybe we can do that if the problem appears more often.
closed 2024-11-05 by hugobuddel