AstarVienna/ScopeSim #516: lms detector ordering

AGlasse, 2024-12-10:
Dear ScopeSim, 
It appears that the LMS detector numbering (from 1 to 4) is ordered in a clockwise direction, rather than left to right.  This may cause complications in the pipeline since the wavelength scale within each detector and for the boresight wavelengths of det 1 to 2 all run left to right, but the boresight wavelengths in dets 3 and 4 are flipped.  Is this a feature?  If it is I'd be grateful to know why you do it this way, 
Thanks,
..Alistair

oczoske, 2024-12-10:
Hi Alistair,
I did it that way because I didn't know any better (I think the numbering is the standard numbering of quadrants in mathematics). In fact, I still cannot find the official layout in the document (I checked version 3 of E-REP-ATC-MET-1003, but may have missed it). Can you provide that, please? Is it just a matter exchanging numbers 3 and 4? For reference, the layout is defined in the file `FPA_metis_lms_layout.dat` in the METIS instrument package (so really an irdb issue), and this is a good time to bring that up to date. The current layout reads
```
id  x_cen    y_cen     x_size  y_size  pixel_size  angle   gain
 1  +19.547  +19.547   2048    2048    0.018       0.0     1.0
 2  -19.547  +19.547   2048    2048    0.018       0.0     1.0
 3  -19.547  -19.547   2048    2048    0.018       0.0     1.0
 4  +19.547  -19.547   2048    2048    0.018       0.0     1.0
```
(coordinate values in mm in the focal plane). Should any of the detectors be rotated or flipped? The gain value obviously needs to be changed (in a development branch I've already set it to 2, which I think is typical for HxRGs).

AGlasse, 2024-12-10:
Hi Oliver,

  No, there is no ‘official’ numbering scheme, I use 1,3 as short wavelength detectors and then number slices 1 to 14 from the bottom to the ‘top’ (gap between 1 and 3) of detectors 3 and 4.  This scheme is then implemented in affine transforms which I use to map from the focal plane at the detector mosaic to the row and column coordinates at the individual detectors.  I have attached my working draft of the distortion report, and a copy of the affine transform fits file to this message for information.   For the extended mode, I am still waiting for our optical engineer to generate ray trace configurations which cover our operational wavelength range,

  Best wishes,
  ..Alistair

Alistair Glasse
JWST-MIRI Instrument Scientist, ELT-METIS CoI
UK Astronomy Technology Centre
The Royal Observatory, Edinburgh
***@***.******@***.***>

UK-(0)-131-385-9399  Zoom
UK-(0)-7887-386-997  Mobile

hugobuddel, 2024-12-11:
A small thing to note is that the pipeline should work irrespective of the ordering of the extensions in the FITS files, because it is not (formally) guaranteed that the Instrument Control Software always writes out the extensions in the same order (although in practice this is the case). The extensions should have headers identifying which is which, so we should identify what these keywords are and what values they should have for the LMS detector.

Alistair, it seems that if you reply be email, that then the text of the issue shows up on the issue tracker (as intended), but any attachment is apparently lost. Could you please attach your working draft to the issue at https://github.com/AstarVienna/ScopeSim/issues/516 manually?

hugobuddel, 2024-12-12:
Reopened because closed accidentally

oczoske, 2024-12-12:
So is there anything that needs to be done about this issue? Note that the numbering of the detectors has nothing to do with the implementation of the EFP-to-MFP transforms, which has always followed the description in E-REP-ATC-MET-1016 and is, I believe, quite similar if not identical to the current draft. Scopesim first computes the (noiseless) MFP  (called `ImagePlane` in Scopesim) and only then sticks in the detectors and reads out. The detector labeling is really quite arbitrary and at this point we only care about where they are in the MFP.

oczoske, 2025-01-31:
I'm closing this as there doesn't seem to be anything that needs to be done right now.

closed 2025-01-31 by oczoske