AstarVienna/ScopeSim #534: Dependabot not quite there yet

teutoburg, 2025-01-27:
Also, more generally with all these dependabot PRs now, I think we're getting into a situation where the versions of various packages in the lock file no longer match the "minimum dependency" versions in the pyproject.toml file. At some point we agreed that this should be kept in sync whenever possible, and we (eventually) figured out how to deal with poetry to achieve that. Now I think dependabot is torpedoing us here...

_Originally posted by @teutoburg in https://github.com/AstarVienna/ScopeSim/issues/533#issuecomment-2616396513_

teutoburg, 2025-01-27:
In addition, it might be useful to make the dependabot group the updates. Like today we got 5 separate ones. Maybe do that weekly or monthly instead, but group? Or not at all and just see how we can get the security updates still? I don't know, but it seems the current configuration is not optimal...

hugobuddel, 2025-01-27:
Yes we should ignore these PRs. I tried to say that in #518 but probably failed to express myself properly:

> Maybe for now keep things as is and manually close these normal PR's until we get a security one to check whether they also happen in the other repos.

My thought process is that these PRs are indeed unwelcome. They are introduced by explicitly adding the `pip` ecosystem (which we did only for ScopeSim).

Without any dependabot configuration we only got the security updates; those are welcome. But it is unclear whether we still get those now we have a dependabot configuration.

So my idea was to keep the `pip` ecosystem in the dependabot configuration for ScopeSim, but not for our other packages. Then we wait until there is an update due to a security problem. Then hopefully the security-PR will show up in all affected projects.

If that security PR indeed shows up for all the projects with a dependabot configuration without the pip ecosystem, then we can remove the pip ecosystem also from the ScopeSim project. If the PR does not show up in other packages, then I don't know what to do.

But that means we are stuck dependabot creating these 'normal' dependency upgrades until we have a security problem. Hence my suggestion to just close them, as we do not want to break our lowest dependencies without good reasons.

teutoburg, 2025-01-27:
> Yes we should ignore these PRs. I tried to say that in [#518](https://github.com/AstarVienna/ScopeSim/pull/518) but probably failed to express myself properly:

Yeah, I may have misread "close" as "merge", without properly thinking about the implications (which only occurred to me today).

> My thought process is that these PRs are indeed unwelcome. They are introduced by explicitly adding the `pip` ecosystem (which we did only for ScopeSim).
>
> Without any dependabot configuration we [...]

Indeed.

> So my idea was to keep the `pip` ecosystem in the dependabot configuration for ScopeSim, but not for our other packages. Then we wait until there is an update due to a security problem. Then [...]

But if we close them, do we get them again the next day? If that's the case, maybe we could set it (here in ScopeSim, for the pip ecosystem) to weekly instead of daily, to avoid the worst of the spam...

teutoburg, 2025-01-27:
I'd still keep the stuff we now merged (semi-unintentionally). I can take care of getting the minimum version back in sync (by updating the pyproject.toml file accordingly).

hugobuddel, 2025-01-27:
I guess things are also different when we split-up ScopeSim in ScopeSim-the-library and ScopeSim-the-application.

Because for ScopeSim-the-library it is not directly our problem if we keep it compatible with dependencies with security issues, so we don't need dependabot for that. (Is that really true? I think so.)

And for ScopeSim-the-application, we can upgrade the dependencies freely, since it is not supposed to be installed together with other tools so we don't care about compatibilities. (As in, ScopeSim-the-application should just be ScopeSim-the-library with optional dependencies like ScopeSim-Targets and jupyter lab etc.)

Maybe I make things too simple. But it would probably help if we split it up.