AstarVienna/ScopeSim #556: Turning off the atmosphere!

AGlasse, 2025-02-10:
Hello Astar.

I am trying to run METIS/LMS simulations for a WCU-like case where I would like to switch off the terrestrial atmosphere.  When I set command parameter

hugobuddel, 2025-02-10:
The default atmosphere for Armazones is the `SkycalcTERCurve` effect called `skycalc_atmosphere`, which fetches the atmospheric spectra from the skycalc server. See https://github.com/AstarVienna/irdb/blob/dev_master/Armazones/Armazones.yaml

This can be turned of by e.g.
```
metis['skycalc_atmosphere'].include = False
```
or
```
simulation.optical_train['skycalc_atmosphere'].include = False
```
depending on how the simulation is initiated (typing from memory).

I believe the pressure is only used for the `AtmosphericDispersion` effect, not for the absolute value of the atmosphere flux.

There is a branch with experimental support for the WCU mode; @oczoske would know more.

AGlasse, 2025-02-10:
Thanks Hugo!  That fixed it...  METIS in space is so much better!  ..Alistair

closed 2025-02-10 by AGlasse