AstarVienna/ScopeSim #568: Filter name is not written back into cmds

teutoburg, 2025-02-17:
```python
>>> sim = Simulation("MICADO")
>>> sim.settings["!OBS.filter_name_fw2"]
'Ks'
>>> sim.optical_train["filter_wheel_2"].current_filter.meta["name"]
'Ks'
>>> sim.optical_train["filter_wheel_2"].change_filter("H")
>>> sim.optical_train["filter_wheel_2"].current_filter.meta["name"]
'H'
>>> sim.settings["!OBS.filter_name_fw2"]
'Ks'
```

This is a problem because other functionality (such as rectification) needs to access the current filter, and also the FITS header get the wrong filter name. Should be addressed in the scope of #561.

hugobuddel, 2025-02-22:
Depends on what we expect to happen.

Say we implement full Observing Block and Template simulations. That is, ScopeSim (or a wrapper) can simulate a full OB with multiple Templates, each with multiple exposures. The "OBS" settings are then specified in the Observing Block. If one of the templates switches to e.g. a Neutral Density filter for a flux calibration exposure, do we then (also) want to keep the original filter in the headers? I don't know the answer, but the original value will be lost if we change the value in `!OBS`.

About the headers, I think headers like ESO.OPTIi.* should actually get the actual value from the effects directly through `#`-strings, and not from the settings through `!`-strings. Maybe the rectification can also use `#`-strings to get the correct value?

teutoburg, 2025-02-22:
I always forget the `#`-strings are a thing. This might be worth a second thought (and a general discussion on how to deal with these kinds of things).