AstarVienna/ScopeSim #608: Bug in extended source flux scaling for spectroscopic modes

teutoburg, 2025-03-26:
I ran into this when trying to re-run an older notebook using METIS imaging and IFU modes, where the difference is something like 10 mag for extended sources. I was able to isolate this using the basic instrument's spectroscopy and IFU modes, see the following tests.

**These two tests pass using ScopeSim v0.8.3, but fail on the current `main`.**

```python
import pytest
import numpy as np
import scopesim as sim

sim.rc.__config__["!SIM.file.local_packages_path"] = "./scopesim/tests/mocks/"

def test_spec_flux():
    src_pnt = sim.source.source_templates.star()
    src_ext = sim.source.source_templates.uniform_source()

    cmd = sim.UserCommands(use_instrument="basic_instrument", set_modes=["imaging"])
    opt = sim.OpticalTrain(cmd)

    opt.observe(src_pnt)
    flux_pnt_img = np.log10(opt.readout()[0][1].data.sum())

    opt.observe(src_ext)
    flux_ext_img = np.log10(opt.readout()[0][1].data.sum())

    cmd = sim.UserCommands(use_instrument="basic_instrument", set_modes=["spectroscopy"])
    opt = sim.OpticalTrain(cmd)

    # Required for v0.8.4
    # opt["source_fits_keywords"].include = False
    # opt["effects_fits_keywords"].include = False
    # opt["config_fits_keywords"].include = False
    # opt["extra_fits_keywords"].include = False

    opt.observe(src_pnt)
    flux_pnt_spec = np.log10(opt.readout()[0][1].data.sum())

    opt.observe(src_ext)
    flux_ext_spec = np.log10(opt.readout()[0][1].data.sum())

    # Compare spec at logarithmic 3 percent level
    assert flux_pnt_spec == pytest.approx(flux_pnt_img, rel=.03)
    assert flux_ext_spec == pytest.approx(flux_ext_img, rel=.03)

def test_ifu_flux():
    src_pnt = sim.source.source_templates.star()
    src_ext = sim.source.source_templates.uniform_source()

    cmd = sim.UserCommands(use_instrument="basic_instrument", set_modes=["imaging"])
    opt = sim.OpticalTrain(cmd)

    opt.observe(src_pnt)
    flux_pnt_img = np.log10(opt.readout()[0][1].data.sum())

    opt.observe(src_ext)
    flux_ext_img = np.log10(opt.readout()[0][1].data.sum())

    cmd = sim.UserCommands(use_instrument="basic_instrument", set_modes=["ifu"])
    opt = sim.OpticalTrain(cmd)

    # Required for v0.8.4
    # opt["source_fits_keywords"].include = False
    # opt["effects_fits_keywords"].include = False
    # opt["config_fits_keywords"].include = False
    # opt["extra_fits_keywords"].include = False

    opt.observe(src_pnt)
    flux_pnt_ifu = np.log10(opt.readout()[0][1].data.sum())

    opt.observe(src_ext)
    flux_ext_ifu = np.log10(opt.readout()[0][1].data.sum())

    # Compare IFU at logarithmic 1 percent level
    assert flux_pnt_ifu == pytest.approx(flux_pnt_img, rel=.01)
    assert flux_ext_ifu == pytest.approx(flux_ext_img, rel=.01)
```

teutoburg, 2025-03-26:
That makes sense, but then I'm even more confused because it _was_ identical pre-v0.9. The actual case where this occurred was the "donut" + star notebook I originally prepared for the METIS meeting last year, and the after v0.9 I suddenly didn't get any flux from the extended part, unless I cranked the `amplitude` (which is interpreted as mag/arcsec²) all the way up from 16 to 1. But that same source then completely saturated the imaging modes, whereas before I could observe the same object in both modes, only with different exptime. So there had to be something up with how the flux is handled for point vs. extended sources in v0.9.

Originally I thought the issue arises from different pixel scales, as I vaguely remember it also occurred for e.g. MICADO (as in, flux was interpreted differently for different scales). But I wasn't able to reproduce that, I'll look into this again later or tomorrow.

hugobuddel, 2025-03-27:
Thanks! The flux conservation confuses me too...

teutoburg, 2025-03-28:
Here's the same notebook (simplified from the original case) saved with output of the two versions in question (as much as I dislike the .ipynb format in other situations, it does come in handy here...). IRDB package versions are identical (never mind the version numbers output by `bug_report()`, those are wrong because the cloned IRDB is used, which has version information in tags, which `bug_report()` doesn't understand yet...), everything else is identical as well. Note that METIS IMG looks the same in both versions, whereas the LMS plot clearly shows the two parts of the "donut" and the central star in the central slice in v0.8, but only the star in v0.9. Interestingly, the MICADO IMG simulation also looks different, not sure if this is the same cause or unrelated.
[ext_flux_demo.zip](https://github.com/user-attachments/files/19507085/ext_flux_demo.zip)

hugobuddel, 2025-03-29:
We really need to get some regression tests in place. Something that we should compare..

This seems to be an excellent case for git bisect, shouldn't be too hard to find the commit that introduced the different behaviour.

teutoburg, 2025-04-05:
Git bisect (my first use of that) revealed that the first commit where the two tests fail is 8a5b219. Now that question is, which is actually correct? That commit seems to have fixed two things: create a copy rather than modify the original data (which was certainly correct and fixed a re-scaling bug when using multiple FOVs I think) and using the pixel area conversion from `cube_cubefields` (I'm honestly not sure about that one right now).

Do we have any photometry tests specifically for extended sources (but not cubes)??

hugobuddel, 2025-04-05:
Honestly, I don't know.

It would perhaps be good to include some sanity checks where we can compute by hand what we expect. E.g. create simple tophat functions for all transmission curves. Sources with a trivial spectrum, etc.

closed 2025-05-27 by teutoburg