AstarVienna/ScopeSim #629: ScopeSim produces negative numbers for MICADO

hugobuddel, 2025-04-06:
ScopeSim produces negative numbers:
```python
import scopesim as sim
import scopesim_templates as sim_tp
import matplotlib.pyplot as plt
import numpy as np
import astropy.units as u
from pooch import retrieve

micado= sim.Simulation("MICADO", mode= ["SCAO", "IMG_4mas"]) #MICADO simulation with SCAO and IMG_4mas mode

# for 9 detectors (if dist from star is &gt;10 arcsec)
for effect_name in ["full_detector_array", "micado_adc_3D_shift", #turn off all effects
                    ]:
    micado.optical_train[effect_name].include = True
    print(micado.optical_train[effect_name])
micado.optical_train["detector_window"].include = False
micado.optical_train["filter_wheel_1"].change_filter("J") #change filter to J
micado.optical_train["filter_wheel_2"].change_filter("open") #change filter to open
micado.settings["!DET.width"]=4096 #set the width of the detector
micado.settings["!DET.height"]=4096 #set the height of the detector

micado.optical_train["relay_psf"].include = False #turn off the PSF
fname = retrieve(
    "<https://scopesim.univie.ac.at/InstPkgSvr/psfs/AnisoCADO_SCAO_FVPSF_4mas_EsoMedian_20190328.fits>",
    known_hash= "a9bd309e5ac025f2aa5f8ded85323465578906e47025ce86501985c9b258474c",
    fname="Anisocado_SCAO_FVPSF_4mas_EsoMedian_20190328.fits",
    progressbar=True,
)
fv_psf= sim.effects.psfs.FieldVaryingPSF(filename= fname, name= "SCAO_FV_PSF")
micado.optical_train.optics_manager["default_ro"].add_effect(fv_psf)

star = sim_tp.star("V", 16 * u.mag, "G2V", x=0, y=0)
primary = sim_tp.star("R", 21 * u.mag, "G2V", x=20, y=0)

stars = star + primary
hdu = micado(stars, 'slac_02_04_2025.fits', dit=4, ndit=30)
data= hdu[4].data
print("Min:", np.nanmin(data), "Max:", np.nanmax(data))
search_data= np.where(data==np.nanmin(data))
plt.plot(data[1880])

plt.xlabel("Pixel")
plt.ylabel("data[1880]")
plt.title("detector hdu[4] [4096:4096], line 1880")
plt.grid(True)

plt.savefig("slac_02_04_2025.png")
```

[image]

[Slack Message](https://scopesim.slack.com/archives/C036RBST0NQ/p1743603060597429?thread_ts=1743595587.320809&cid=C036RBST0NQ)

oczoske, 2025-04-06:
The obvious culprit is the `FieldVaryingPSF`. The readout image (all detectors) has a dark grid imprinted on it and the negative pixels presumably all lie within that grid. With the `FieldConstantPSF`, everything is positive (I don't think I see the stars, though).

hugobuddel, 2025-04-07:
Oh yes of course that must be it. I didn't properly look at the resulting FITS file; I couldn't see it with the contrast I used.

For some reason I still like skycat for such things, should try to get that installed on this laptop as well.