AstarVienna/ScopeSim #644: Conflicting definitions of `__getitem__` in `Effect` and `SpectralTraceList`

oczoske, 2025-04-21:
While trying to access certain parameters from `MetisLMSSpectralTraceList` for inclusion in FITS headers (see https://github.com/AstarVienna/irdb/pull/229) I noticed that they could not be accessed via the `#lms_spectral_traces.wave_min` mechanism. It turns out that this mechanism relies on the rather elaborate definition of `Effect.__getitem__`, which accesses `Effect.meta`. This is, however, masked by `SpectralTraceList.__getitem__`, which returns a particular `SpectralTrace` (implemented in  https://github.com/AstarVienna/ScopeSim/pull/229, which ironically starts with "This should be harmless."). 
From a practical point of view it might be easier and less effort to remove `SpectralTraceList.__getitem__` (although it's a quite natural use for what is, after all. essentially a list). The less natural definition in `Effect` is presumably harder to replace (I don't have a good overview of where it is used), but might cause further confusion in the future.
The discussion in https://github.com/AstarVienna/ScopeSim/issues/387 is probably relevant (but goes over my head).

teutoburg, 2025-04-22:
Removing the `__getitem__` from `Effect` makes 10 tests fail (less than I expected). All of those seem to come out of `OpticalElement.__getitem__()` which redirects to the `Effect.__getitem__()` if given a `#`-string. TBH, `OpticalElement.__getitem__()` seems a bit weird to me, or rather inconsistent, you might call it "smelly". I also realized I don't yet have a full grasp of the interplay between `Effect` (and it's various subclasses), `OpticalElement` and `OpticsManager`.

Either way, it seems the `Effect.__getitem__()` is only actually used in that line in the `OpticalElement` (or any other use is untested). I'd be fine with replacing this questionable use of `__getitem__` (I don't think an `Effect` _is_ anything like a list of a dict outside of special ones, like, you know, `SpectralTraceList` and such) with a simple method (name TBD) that does the same thing. Maybe then `OpticalElement` could receive the same method and the functionality of `OpticalElement.__getitem__()` could be split, but that can be discussed separately.

oczoske, 2025-10-06:
Bumping this issue. The parameters from MetisLMSSpectralTraceList need to go into FITS headers soon-ish.