AstarVienna/ScopeSim #723: Move flux binning to `ImagePlane`?

teutoburg, 2025-07-03:
The `ImagePlane` should only store binned flux values (i.e. photons per second). In between (i.e. in and around the FOV), we (_sometimes_ in imaging, _always_ in spectroscopy) need to work with differential units. The location where the binning is done varies:

- In imaging, this is done inside the `FieldOfView.make_xy()` methods for each field.
- In normal spectroscopy, this is done somewhere in the depths of the spectral trace classes.
- In simple IFU mode, this is done in the separate `FluxBinning3D` effect (because we never go to the spectral traces).

Think about the last case again, I'm unsure if it was a good idea to create that effect or not. There appears to be an alternative, which is to do this inside `ImagePlane.add()`. Especially once the flattening is moved there (see #721), the current (rudimentary) BUNIT checks there might be expanded to catch incoming non-binned data and bin it right there and then, before adding to the `ImagePlane`.

But that also goes into the whole discussion of #692. We (all) should at some point make a Grundsatzentscheidung, whether we want to move all "physical type changes" (see also [the new flowchart](https://scopesim.readthedocs.io/en/stable/faqs/scopesim_units_flow.html)) into Effects (meaning a yes to #692, but a no to this here), or rather keep these things mostly in `optics` classes (the other way around).

I suggest discussing this between all of @AstarVienna/scopesim-devs after Leiden.

hugobuddel, 2025-07-04:
My personal (current) conclusion is that differential units are ultimately superior, so if we need a principebesluit, then I'll suggest to keep differential (spatial) units as long as feasible. For me that means putting the binning in an Effect, because 'as long as feasible' depends on the instrument/mode/simulation.

The prime reason differential units are superior is because they allow an arbitrary and flexible resolution that is (or can be) decoupled from the instrument specifics. Also, differential units are a strict superset of binned units, as one can select the unit to be exactly the size of the bin.

A practical reason to keep differential flux units as long as possible is that the pixel are not actually all exactly the same size. Their sensitivity is also not uniform: the sensitivity drops at the edges.

Instruments like MICADO aim to have at least 1/30th of a pixel sub-pixel accuracy, e.g. in centroiding. Sensitivity differences within the pixel will be relevant for those instruments, but only for certain pixels (e.g. those with stars, depends on science goals). Defaulting to differential flux units would give us a basis on which we can build support for such adaptive sub-pixel precision.

I also envision a future in which an analytically described source (e.g. a stellar cluster) can be kept analytical as long as possible. E.g. by applying an analytical PSF, analytical ADC, etc. An analytical source almost implies differential units.

What this means practically (at least to me) is that we should do the binning as late as feasible, and explicitly. So not somewhere "hidden in the depths", but as an explicit step. Maybe we even want to be able to move that step around based on the desired simulation tradeoffs (e.g. speed vs accuracy vs memory consumption), which would be easier with an Effect. 

So having the binning in an Effect sounds like a logical choice to me.

teutoburg, 2025-07-04:
Initially, I used to prefer binned units, because something like "I can just count these photons that come in each second" is somewhat easier to think about. But I have since learned the advantages of differential untis, so I also think those should be preferred. In the FOV, this is already done in the cube case (and also in the incoherent spectrum case which isn't actually used), and I'm in the process of also using differential units for the imaging case (_without_ any messy `use_photlam` flags or what not), which is also partially where this question came up now.

But those are really two separate discussions I think. As in, we could totally e.g. keep differential units in the `FieldOfView` and still do the binning in the `ImagePlane` (as opposed to in an effect). But being able to "move around" the exact point where the binning occurs sounds interesting (although it might be hard to do in practice, because of the inherent split in where each effect is applied (i.e. Source/FOV/ImagePlane/Detector/etc.)).

Regarding keeping parametrized sources non-binned as long as possible, that's also something I have in mind for scopesim-targets. My current plan is to at least do that until they hit the `OpticalTrain` (I'm not yet sure which way around exactly that should happen, but the basic idea is that you could take the same `Target` object and chuck it into different `OpticalTrain`s (and perhaps get two different `Source` objects, but that's TBD)), somewhat similar to what that `prepare_source` method does to cubes. But with keeping differential units around for longer, this might be able to be shifted further down.