AstarVienna/ScopeSim #724: Create table source directly from file

teutoburg, 2025-07-03:
Currently, it is possible to pass a `filename` and `spectra` to `Source`, where `filename` points to either a FITS file with a BINTABLE extension (not the primary) or a ASCII file. But the `spectra` still need to be given in some manual form. It would be nice to have the functionality to also store one or more spectra in additional BINTABLE extension(s) and reference those, such that a table-type `Source` could be created directly from one FITS file.

As a future bonus, we might even allow for ImageHDU(s) _and_ point source tables plus all their spectra in a single FITS file, that gets turned into a composite `Source` object. Check how the first part of this (images and their spectra) is implemented in the infamous spiral two component source templates...

hugobuddel, 2025-07-04:
Maybe turn things around and think about what it would mean for `Source` class to have a `save_to_fits()` method?

At least conceptually; because that mindset might clarify how such a FITS file should look in the most general case. And once we have thought about it, then we might as implement it 😬 .

Or a step further: I'd like to have the headers of the resulting FITS file to contain all the parameters to redo the exact same simulation (and thus recreate the exact same pixels). E.g. in the `HIERARCH SIM` namespace. Currently, the Source is missing in that scheme; I recall we hijacked the `TARGET` keyword to just contain a ScopeSim_Templates function call. But maybe we could (optionally) add this hypothetical FITS file as extensions to the simulated output.

teutoburg, 2025-07-04:
For scopesim-targets, I always had the vision to be able to define them via a yaml file (for anything that isn't an image or spectrum or even cube, rather the more simple parameters). But I guess those two options are not mutually exclusive. As in, we can just support both FITS headers and yaml files for that kind of scalar data (key-value(-comment) kind of data, really, in database lingo). Any spectra/images/cubes (I was gonna call those "binary", but especially spectra we usually store as ASCII, not binary, but in db terms probably still a "blob"), let's call it vector-ish (tabular?) data, will still be stored in either a FITS (images, cubes) or FITS/ASCII (spectra) file (referenced in the yaml) or a FITS extension (in the case that the scalar data is stored in a FITS header).

Side-rant: We should at some point decide on a (at least internal) standard representation of spectra in FITS files. ASCII is easy enough, but for FITS, my experience is that every project does this somehow slightly different...