AstarVienna/ScopeSim #757: `utils.find_file` does not check for files in current working directory

oczoske, 2025-08-13:
### What happened?

Sometimes it is useful to override files in the irdb/`inst_packages` with self-made files that are typically in the current working directory or elsewhere in the user's system. An example would be
```
metis['wcu_sources'].set_fpmask("my_lovely_fpmask.dat")
```
or
```
metis['slit_wheel'].change_slit("./I_want_to_test_this_slit_design.dat")
```
The search path typically does not include `./` (why not?) and if it is to be that way then one could argue that the first form should be indeed be restricted to the search path. However, the second form (any kind of relative path, really) is very explicit about which file the user wants to use. `utils.find_file` only tests for absolute paths, otherwise it goes directly into the search path.
Any reason why there shouldn't be a simple test for `filename.exists()`? 

### Code to reproduce

```python

```

### Exception traceback

```bash

```

### ScopeSim bug_report()

```bash
Python:
3.12.9 (main, Feb  5 2025, 01:31:18) [GCC 14.2.0]

Installed Python packages:
           anisocado: 0.3.0
         astar-utils: 0.3.3
             astropy: 6.1.7
      beautifulsoup4: 4.13.4
            docutils: 0.19
               httpx: 0.28.1
                lxml: 5.4.0
          matplotlib: 3.10.5
      more-itertools: 10.7.0
               numpy: 1.26.4
           packaging: 25.0
               pooch: 1.8.2
              pyyaml: 6.0.2
               scipy: 1.16.1
            scopesim: 0.10.0
       scopesim_data: could not be loaded.
  scopesim_templates: could not be loaded.
         skycalc-ipy: 0.5.3
             synphot: 1.6.0
                tqdm: 4.67.1
```

### IRDB installation

Downloaded via sim.download_packages()

### Use case

Other

hugobuddel, 2025-08-13:
Don't know, sounds fine to me to add the current working directory to the search path. As the first or the last element of the search path?

teutoburg, 2025-08-13:
I never took the time to fully understand how the search path and find file interact. But it shouldn't be too hard to try adding the CWD both in first and in last place, and see what happens...

closed 2025-08-25 by oczoske