AstarVienna/ScopeSim #797: PSF cube in MICADO spectroscopy has undesirable features

oczoske, 2025-09-17:
The image shows cuts through a trace on the MICADO ImagePlane, source is a star, the slit is `SPEC_3000x48`. The blue curve shows the simulation with the default PSF in the irdb, `PSF_SCAO_ConstPSF_0_5off.fits`, which has an image size of 256x256 pixels. The orange one uses a PSF of size 1024x1024. There are two features that are undesirable:
- The small PSF sits on a constant pedestal (different values left and right) that extend from the edge of the PSF over a total extent of 512 pixels. This probably arises from extrapolation using a constant value set by the edge pixels of the PSF image. This pedestal needs to be lowered to 0. Since `RectBivariateSpline` has no parameter to control the extrapolation (why, oh, why?), we probably have to set the edge of the PSF image to zero.
- The large PSF is cut to 512 pixels, which makes gives the star spectrum an obvious sharp edge in the `ImagePlane`. Depending on the brightness of the star this may or may not disappear in the noise in a readout. 

Both features are generated in `make_psf_cube` in `psfs/discrete.py`. Cutting to 512 is hardcoded in
```
# adapt the size of the output cube to the FOV's spatial shape
nxpsf = min(512, 2 * nxfov + 1)
nypsf = min(512, 2 * nyfov + 1)
```
In our case we have `nxfov=750`, hence the PSF is cut to 512. The question is what to replace the 512. The size of the input psf is one option, but that may lead to huge memory requirements. A user-settable parameter is probably needed.

teutoburg, 2025-09-17:
> The image shows

Should there be an image attached? I don't see one

oczoske, 2025-09-18:
Huh. I kept telling myself "Don't forget to attach the image"... Added it now.

PR #798 (2025-09-20): Use `RegularGridInterpolator` for proper extrapolation of small PSF images
This fixes #797 .
- The interpolator has been changed from `RectBilinearSpline` to `RegularGridInterpolator`. While the interface is more cumbersome it has better support for extrapolation - here we need `fill_value=0`. We still use linear interpolation but might consider using `pchip`; tests are needed.
- The spatial size of the scaled PSF cube is at most twice the slit size (ensures full coverage of the slit by the psf), otherwise it takes the size of the input PSF image. This can be further reduced by setting the par [... cut]
closed 2025-09-20 by oczoske